HKUST-Aerial-Robotics/VINS-Fusion #1: synchronized stereo camera and IMU needed? (using zed_stereo camera and MPU9250)

FrozenBurning, 2019-01-16:
Considering my experience in VINS-Mono, synchronized device in hardware is needed. but this time, in VINS-Fusion, I face the same drifting problem as I have faced in previous time. do you have any suggestion in hardware (since I use zed stereo camera and MPU9250 ,not well synchronized)?

pjrambo, 2019-01-16:
> > > Considering my experience in VINS-Mono, synchronized device in hardware is needed. but this time, in VINS-Fusion, I face the same drifting problem as I have faced in previous time. do you have any suggestion in hardware (since I use zed stereo camera and MPU9250 ,not well synchronized)?
> > 
> > 
> > You may try to set the value "estimate_td" to 1 in the config file.
> 
> I have set it to '1', but the online estimation didn't work well. : (

As it is said in the Readme, "VIO is not only a software algorithm, it heavily relies on hardware quality. For beginners, we recommend you to run VIO with professional equipment, which contains global shutter cameras and hardware synchronization."
If you want to get a better result with your own devices, hardware synchronization is needed.

Here are two options:
1. If you are unfamiliar with hardware synchronization, you'd better use professional equipment, like some visual-inertial modules.
2. Do hardware synchronization for your own devices.

FrozenBurning, 2019-01-17:
> As it is said in the Readme, "VIO is not only a software algorithm, it heavily relies on hardware quality. For beginners, we recommend you to run VIO with professional equipment, which contains [...]

Fine, I will do the hardware synchronization and try again! 
Thanks.

closed 2019-01-17 by FrozenBurning