HKUST-Aerial-Robotics/VINS-Fusion #3: the global optimization thread doesn`t work properly

AIgithub22, 2019-01-17:
I test vins-fusion with my own device (global shutter camera + imu + gps(RTK) put those together but not hardware synchronized) and set estimate_td to 1, online estimate time offset between camera and imu.
it seems that the global optimization thread doesn`t work properly， at most time the global optimized path overlay exactly with the vio odometry path, however the gps(RTK) path was not overlay with those two path. 
And sometimes the global optimized path jumps  for a short distance and ran off very fast , but the gps and vio path  still works fine.
I could sent my rosbag to you if you are interested. 
[image]

pjrambo, 2019-01-17:
> I test vins-fusion with my own device (global shutter camera + imu + gps(RTK) put those together but not hardware synchronized) and set estimate_td to 1, online estimate time offset between camera [...]

According to your description, I guess the unsynchronized (imu+camera) and GPS may cause that problem. If your gps timestamp has a big time delay, at the beginning the estimator can not get the corresponding gps data, so your vio path may be same with the global path. Once the first time the estimator get the gps data, because of the unsynchronized data, it may optimize vio data with wrong gps data, which may lead to the issue as you said. 
If this is not the reason for your case, you may send me your bag and config file. You can send it to jpanai@connect.ust.hk or just attach the link below.

zjlooojoe, 2019-05-07:
Can you tell me where the dataset you get?Because I want to use imu+camera+gps too,but I cant find a good dataset.KITTI has the three,but the rate of imu is same with gps and camera.Maybe the kitti's unsynchronized data is what I want?

HpyZhai, 2019-05-10:
I have the same problem as you. Did you solve it?

zjlooojoe, 2019-05-11:
https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/issues/39
Do you mean that?

HpyZhai, 2019-05-13:
No, I mean the fusion path is very similar to the vio odometry  path.  The reason is that my gps timestamp is not the same as the vio corresponding timestamp. So the gps position constrain can't work. The code in line 160，in the globalOpt.cpp causes the problem.

yichengethanli, 2021-12-10:
> I test vins-fusion with my own device (global shutter camera + imu + gps(RTK) put those together but not hardware synchronized) and set estimate_td to 1, online estimate time offset between camera [...]

Hi,
I am currently working on a similar project that I would like to fuse GNSS/RTK data with VIO. Im quite interested in your experiment and wonder if I can take a look at your dataset. Could you send your rosbag dataset? My email is lyc0732@hotmail.com. Please feel free to contact me. Thanks.