HKUST-Aerial-Robotics/VINS-Fusion #4: Dose the node loop_fusion only works with IMU?

ZHAORANWU1, 2019-01-17:
I have tried VINS-FUSION with kitti-dataset with stereo mode without IMU.and I find that the path of the result will not processed by Loop_fusion node evev i have run it.

I have run it like this:
 roslaunch vins vins_rviz.launch

rosrun loop_fusion loop_fusion_node /localhome/proprak8/catkin_ws/src/VINS-Fusion/config/kitti_odom/kitti_config00-02.yaml 

rosrun vins kitti_odom_test /localhome/proprak8/catkin_ws/src/VINS-Fusion/config/kitti_odom/kitti_config00-02.yaml /localhome/proprak8/Dataset/dataset/sequences/00/

pjrambo, 2019-01-18:
> I have tried VINS-FUSION with kitti-dataset with stereo mode without IMU.and I find that the path of the result will not processed by Loop_fusion node even i have run it.
> 
> I have run it like this:
> roslaunch vins vins_rviz.launch
> 
> rosrun loop_fusion loop_fusion_node /localhome/proprak8/catkin_ws/src/VINS-Fusion/config/kitti_odom/kitti_config00-02.yaml
> 
> rosrun vins kitti_odom_test /localhome/proprak8/catkin_ws/src/VINS-Fusion/config/kitti_odom/kitti_config00-02.yaml /localhome/proprak8/Dataset/dataset/sequences/00/

The loop_fusion node only support mono+imu and stereo+imu now. It does not support pure stereo mode.

qintonguav, 2019-01-20:
@ZHAORANWU1  For your information, I add loop closure function to VO mode. Thanks @pjrambo

ZHAORANWU1, 2019-01-23:
Thanks for the addtion! 
We have tested the Loop-fusion node with stereo mode.It works now

liangsongLee, 2019-03-14:
Hello, how can I get the txt file of the trajectory that has been loop closed when I use only kitti stereo mode? I can only get the .txt before loop closing in the output folder. I see that the loop data is not in the form of txt, it is in the form of csv.  How can I use it? Thank you very much! @qintonguav

liangsongLee, 2019-03-17:
Hello, can you get the pose after loop closure using stereo  mode? @ZHAORANWU1

20221023lijia, 2024-08-06:
> Hello, can you get the pose after loop closure using stereo  mode? @ZHAORANWU1 

hello! Did you solve the problem?i can't get the pose_graph photo(=ＴェＴ=)  it's empty(。•ˇˍˇ•。)