HKUST-Aerial-Robotics/VINS-Fusion #10: Vins stereo problem

LittleByLittle2, 2019-02-07:
Thank you for your great work. And I can run the stereo vins with Kitti data set perfect. But when I ran it with my own data set, the trajectory is bad. 

```
roslaunch vins vins_rviz.launch
rosrun vins kitti_odom_test ~/catkin_ws/src/VINS-Fusion/config/kitti_odom/my_kitti.yaml My_DATASET_FOLDER/sequences/seq25/ 
```

The dataset is from parking lot.

I already changed the camera calibration file and time stamp has the same unit as kitti. Besides, I saw the feature can track well. 

So I am wondering what can cause this issue?  And I can not see the parameter setting of baseline of stereo camera in the setting file. Where is it?

johnsongash, 2019-02-17:
Same question here,
how can we set the baseline of stereo without imu?
The result for pure stereo VO seems poor...

qintonguav, 2019-02-17:
@johnsongash set identity to cam0. Use some toolbox (such as opencv stereo) to calibrate the transform between two cameras. Write this transformation to cam1. Please refer to KITTI dataset example.

LittleByLittle2, 2019-02-18:
> @johnsongash set identity to cam0. Use some toolbox (such as opencv stereo) to calibrate the transform between two cameras. Write this transformation to cam1. Please refer to KITTI dataset example.

OK, thank you. Got it. But another question is the tracking method is light flow method, this kind of method has the assumption that the photometric of same pixel is stay the same, so have you tried the method in the light changed faster scenario?

ZW-PRO, 2024-02-17:
> 将 Identity 设置为 CAM0。使用一些工具箱（例如 opencv stereo）来校准两个摄像机之间的转换。将此转换写入 cam1。请参考 KITTI 数据集示例。

你好，我将cam0设置为单位矩阵，cam1设置为[R|T]，但是漂移严重，不知道问题出在哪里