HKUST-Aerial-Robotics/VINS-Fusion #13: Is there value in using VINS-Mono anymore or does this supercede in every meaningful way?

SteveMacenski, 2019-02-08:


tuandle, 2019-02-09:
I think this is explained in the paper. Stereo version has worse accuracy performance than Mono version because of inaccurate calibration and IMO, feature tracking using stereo camera consumes more time and leaves a little time for optimization. I never do any benchmarking so this is purely my speculation. On the other hand, 3d triangulation with stereo is precise (given accurate calibration) so you don't have to always move around to get enough parallax for triangulation as in VINS-Mono. 
In general, it depends on your application. If you use a drone, VINS-Mono is still the best choice (lightweight, less power consumption, no worry about the parallax thing,etc). But for a ground vehicle, stereo is preferable.

SteveMacenski, 2019-02-10:
It looks like VINS-Fusion will allow you to use mono + IMU like VINS-Mono, not just stereo. If the issue is just in the general practice of using stereo that you point out, wouldn't the mono + IMU version of VINS-Fusion be comparable to VINS-Mono?

tuandle, 2019-02-10:
I probably confuse myself between (VINS-Mono vs VINS-Fusion) and (mono vs stereo). You are right. What I said earlier was about mono vs stereo case. Checking the config file for camera, I think they introduce forward and backward optical flow which doesn't exist in VINS-Mono. Rather than that, I believe the rest is the same. 
@qintonguav @pjrambo may answer this best :)

wlsh24, 2019-02-12:
Another change is how loop closure is handled (see my question here: [issue 8](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/issues/8)).

In VINS-Mono, there are some constraints added from the relocalization node [estimator.cpp#L769](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/vins_estimator/src/estimator.cpp#L769).
While in VINS-Fusion, the relocalization node is completely independent from the estimator (you can check the optimization [here](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/master/vins_estimator/src/estimator/estimator.cpp#L926)).

I am not exactly sure why this change was made since I believe adding the constraints on the matched tracks should improve the odometry result (VINS-Mono).
I hope that the authors can help us understand the difference.

mmp52, 2020-01-16:
@wlsh24 I also have some issues on loop closure, have you sold this point yet? Do you suggest to use VINS-Mono instead of Fusion when working with a single camera+imu?

thanks

arjunbj, 2020-12-30:
Hi, bringing this thread up again... did anyone have a chance to get better insights about this? Wondering which would be better to try for ground bot.