HKUST-Aerial-Robotics/VINS-Fusion #14: wechat for VINS-Fusion

nonlinear1, 2019-02-11:
I make a team of wechat for VINS-Fusion, This team can discuss all kinds of issue of vins and other vio slam.
If anyone want to join it. Please ,first add wechat number:31781410.
我建了一个微信群供大家一起讨论VINS-Fusion以及其它vio slam, 如果大家想要加入可以先加一下微信号31781410

HKVINS89757, 2019-06-11:
已加

antoan, 2019-06-23:
Hi, I'd really like to join, however as I am based in the UK , to register I need an Wechat friend as a reference to get through the registration process, after installing the app. 

I was wondering if somebody would be kind enough to help me with this.

HKVINS89757, 2019-06-23:
> the

> Hi, I'd really like to join, however as I am based in the UK , to register I need an Wechat friend as a reference to get through the registration process, after installing the app.
> I was wondering if somebody would be kind enough to help me with this.

wechat ID： yuluoxijie

antoan, 2019-06-23:
@HKVINS89757 Thank you so much.
I will try to register later tonight.

antoan, 2019-06-23:
@HKVINS89757
The App tells me to ask my WeChat friend to open their WeChat app and navigate to: 

Contacts -> Search "WeChat Team -> Account -> Help Friend Register

Then enter my number: +44 7305858042

I hope that is ok.

nonlinear1, 2019-06-24:
You should search the account:31781410. This is my account and then I invite you join our team!

YuanwenFu, 2019-09-29:
I added the wechat number:31781410 just now!

YuanwenFu, 2019-09-29:
And I hope you agree!