HKUST-Aerial-Robotics/VINS-Fusion #16: MYNTEYE-D have large delay in stereo+imu mode

shmm91, 2019-02-18:
Thank you for your great work.
My device is MYNTEYE-D, the image size is 1280*720. When I move the device, the track image and odometry in RVIZ have large delay. For example, I hold the device move around, the track image and odometry needs 5~10 second to respond. Do you know why?

arjunskumar, 2019-02-18:
@shmm91 I had the similar problem with Myteye and Intel D435

shmm91, 2019-02-18:
@arjunskumar Have you solve the problem?

arjunskumar, 2019-02-22:
@shmm91 not yet, i have this delay issues with Tx2 and intel realsense D435

arjunskumar, 2019-02-26:
@shmm91 try reduce the feature count and turn off flow back. I got it working on my pc but not in TX2

Lip0, 2019-04-01:
> @shmm91 not yet, i have this delay issues with Tx2 and intel realsense D435

pamzerbhu, 2019-04-04:
Hi, the same device MYNTEYED and tx2 that I have met the same question also try to use VINS-Fusion-GPU but the problem remains.

0Jiahao, 2019-04-04:
@arjunskumar Hi! Have you tried out the VINS-Fusion-GPU? I can only run it by reducing features (to 80) and disabling the flow_back. Is it same to yours?

pamzerbhu, 2019-04-06:
> Hi, the same device MYNTEYED and tx2 that I have met the same question. I try to use VINS-Fusion-GPU but not help with the problem.

I have solved it by setting flow_back: 0 max_cnt: 70  and use VINS-Fusion-GPU.

Lsmartworld, 2019-05-15:
Hello, the same device MYNTEYED and tx2 that I have met the anther question.During the process of using VINS-FUSION, I found that when running stereo + IMU, the drift is very powerful. But when running stereo, it is very stable. Would you like to ask if you have encountered such a problem? @pamzerbhu

pamzerbhu, 2019-05-19:
> Hello, the same device MYNTEYED and tx2 that I have met the anther question.During the process of using VINS-FUSION, I found that when running stereo + IMU, the drift is very powerful. But when [...]

I am sorry hadn't use stereo mode,  I only tried to use mono+IMU. But I have two suggests for this problem.
 1. Check you Extrinsic Parameters between IMU and Cam.  The standard Extrinsic Parameters which you read from Cam by MYNT SDK.
2. The scale of IMU parameters read by MYNT SDK has been enlarged 100 times. Please check it.
 Pamzerbhu

Lsmartworld, 2019-05-19:
> I am sorry hadn't use stereo mode, I only tried to use mono+IMU. But I have two suggests for this problem.
>
> 1. Check you Extrinsic Parameters between IMU and Cam.  The standard Extrinsic Parameters [...]

Thank you for your answer.I can't understand the second suggest. where can check the parameter?

pamzerbhu, 2019-05-20:
> Thank you for your answer.I can't understand the second suggest. where can check the parameter?
Let me see files called:
1. "MYNT-EYE-VINS-FUSION-Samples/config/mynteye/mynteye_stereo_imu_config.yaml" 
2.  Param file you read from Cam named "image_params.params" and "imu_params.params"

Lsmartworld, 2019-05-20:
> 1. "MYNT-EYE-VINS-FUSION-Samples/config/mynteye/mynteye_stereo_imu_config.yaml"
> 2. Param file you read from Cam named "image_params.params" and "imu_params.params"

Thank you very much ! But I can't solve the problem.

zxh2015, 2019-09-17:
Have you solve the problem?

pamzerbhu, 2019-09-17:
I have solved it by setting flow_back: 0 max_cnt: 70 and use VINS-Fusion-GPU.

> Have you solve the problem?

I have solved it by setting flow_back: 0 max_cnt: 70 and use VINS-Fusion-GPU.

zxh2015, 2019-09-18:
> I have solved it by setting flow_back: 0 max_cnt: 70 and use VINS-Fusion-GPU.
> 
> > Have you solve the problem?
> 
> I have solved it by setting flow_back: 0 max_cnt: 70 and use VINS-Fusion-GPU.

oh, no, I don't have GPU with my computer....but I will try it. Thanks!

zxh2015, 2019-09-18:
I have solved it by starting the SDK with another launch file
`
roslaunch mynteye_wrapper_d vi_fusion.launch streams_mode:=1
`
now, the image size is 1280*480
I don't know  if the size is the cause .
it works

closed 2020-04-29 by shmm91

robin-shaun, 2021-03-20:
Reducing the frame rate works.