HKUST-Aerial-Robotics/VINS-Fusion #17: Kitti GPS Mode Results are different with Video

LeoKingBUAA, 2019-02-19:
Hi，

I tested VINS-Fusion in Stereo+GPS mode with Kitti dataset, but the results are different from video presentation. Differences are in these aspects:
1. There are two trajectory in my test, but only one in video.
2. In my test the trajectory fused with GPS don't have map features, but the trajectory in video have map features.

So I wan't to ask if I config something wrong or there is another version that you don't open source?

animeshsahu80, 2019-05-28:
@LeoKingBUAA 
 did you find any solution?

zxh2015, 2019-10-08:
Have you solved this problem?

zxh2015, 2019-10-09:
> @LeoKingBUAA
> did you find any solution?

I found this to be correct, there will be a yaw angle between the trajectories.

pauls0323, 2024-01-12:
@zxh2015 有办法把这个两个轨迹对齐吗？