HKUST-Aerial-Robotics/VINS-Fusion #19: Issue with running Euroc MAV dataset (MH03)

YuCodes, 2019-02-19:
I tried to run the vins FUSION as per the instructions for, but it get an error after like 2 seconds and it stops. basically it mentions "STACK SMASHING DETECTED"

[ INFO] [1550620405.682573881]: Initialization finish!
time: 1403637131.588319, t: -0.000260 -0.000035 -0.000052 q: 1.000000 -0.000010 0.000556 0.000094 
time: 1403637131.688319, t: -0.000228 -0.000094 0.000004 q: 1.000000 0.000311 0.000445 0.000057 
time: 1403637131.788319, t: -0.000063 -0.000040 -0.000013 q: 1.000000 0.000191 -0.000131 -0.000072 
time: 1403637131.888319, t: -0.000049 -0.000002 -0.000035 q: 1.000000 -0.000036 -0.000057 -0.000011 
time: 1403637131.988319, t: -0.000071 -0.000014 -0.000039 q: 1.000000 0.000157 -0.000170 -0.000254 
time: 1403637132.088319, t: -0.000004 0.000096 -0.000015 q: 1.000000 -0.000152 -0.000054 -0.000741 
time: 1403637132.188319, t: 0.000017 0.000139 -0.000030 q: 1.000000 -0.000232 -0.000095 -0.000753 
time: 1403637132.288319, t: 0.000013 0.000144 -0.000021 q: 0.999999 -0.000100 -0.000136 -0.001037 
time: 1403637132.388319, t: 0.000000 -0.000021 0.000008 q: 1.000000 0.000114 0.000089 0.000028 
time: 1403637132.488319, t: -0.000015 0.000106 -0.000061 q: 1.000000 -0.000116 -0.000059 -0.000776 
time: 1403637132.588319, t: 0.000051 0.000258 -0.000015 q: 0.999999 -0.000300 -0.000106 -0.001571 
time: 1403637132.688319, t: 0.000045 0.000088 0.000027 q: 1.000000 -0.000144 -0.000043 -0.000530 
time: 1403637132.788319, t: 0.000091 0.000041 0.000068 q: 1.000000 -0.000068 0.000075 -0.000344 
time: 1403637132.888319, t: 0.000178 0.000132 0.000071 q: 1.000000 -0.000209 0.000352 0.000138 
time: 1403637132.988319, t: -0.000112 0.000046 0.000033 q: 0.999999 -0.000398 0.000783 0.001066 
time: 1403637133.088319, t: 0.000195 -0.000014 -0.000074 q: 0.999999 -0.000514 0.000976 0.000640 
time: 1403637133.188319, t: 0.001971 0.000117 -0.002725 q: 0.999996 0.000633 0.002543 0.001089 
time: 1403637133.288319, t: 0.004889 0.001171 -0.004884 q: 0.999984 -0.003344 0.001760 -0.004173 
*** stack smashing detected ***: /home/yuthika/vins-ws/devel/lib/vins/vins_node terminated
Aborted (core dumped)

pjrambo, 2019-02-20:
> I tried to run the vins FUSION as per the instructions for, but it get an error after like 2 seconds and it stops. basically it mentions "STACK SMASHING DETECTED"
>
> [ INFO] [1550620405.682573881]: [...]

What kinds of eigen version are you using? It seems that latest eigen version may cause this issue.
I had met a similar issue when I installed eigen 3.3.7 on my computer. After I uninstall that latest eigen and use my previous version of eigen, it works.

YuCodes, 2019-02-20:
I'm using eigen 3.3.5, i'll downgrade and let you know. 
Thanks for the info

closed 2019-02-20 by YuCodes