HKUST-Aerial-Robotics/VINS-Fusion #25: Eigen Unaligned Array Assert error

neorobo, 2019-03-13:
I've been running VINS-Fusion on a standard intel laptop, but want to try it on a raspberry pi 3 b+ and see what I can do through tuning low feature tracks and low speeds/framerates. I've compiled it on the pi, but when I attempt to run the same configuration as what is on my intel laptop, I get the error described here: https://eigen.tuxfamily.org/dox/group__TopicUnalignedArrayAssert.html

The Eigen version that's installed on the laptop is 3.2.92 and the one on the raspberry pi is 3.3.2. The error seems to be coming from Estimator::processImage. Is this something that needs to be fixed according to the above link? Or is there a recommended version of eigen that I should be using?

shaozu, 2019-03-14:
Hi, @neorobo. We use eigen 3.3.3 on TX2, which also has an arm-architecture CPU. BTW, you can try our docker version to see if it can work.

moaljazaery, 2019-07-03:
I solved the issue by changing CMakeLists.txt in both (camera_models) and (vins_estimator) folders.
**Update this line :** 
set(CMAKE_BUILD_TYPE "Release")
**to:**
set(CMAKE_CXX_FLAGS "-std=c++11 -O3 -Wall -g -march=native")

**Also use** CERES_VERSION="1.12.0" and eigen 3.3.3