HKUST-Aerial-Robotics/VINS-Fusion #26: get cam0_calib file

zz1120122573, 2019-03-15:
Hi, if I use another stereo camera, how can I transform
cam0_pinhole.yaml to  cam0_mei.yaml?  

Thanks!

shaozu, 2019-03-15:
Hi, pinhole itself is ok, and there is no need to transform one to the other.

closed 2019-03-15 by shaozu

volcanoh, 2019-05-08:
> Hi, pinhole itself is ok, and there is no need to transform one to the other.

I have a question, why pinhole is worse than mei or equi is euroc MH01 , but pinhole is ok in vins mono?

zhouhaoran-TJU, 2020-10-11:
> > Hi, pinhole itself is ok, and there is no need to transform one to the other.
> 
> I have a question, why pinhole is worse than mei or equi is euroc MH01 , but pinhole is ok in vins mono?

Hi, I have the same question, It's really puzzling that when I using pinhole.yaml, it performed worse, although the camera type is pinhole in the dataset.