HKUST-Aerial-Robotics/VINS-Fusion #28: Can stereo+imu version be used in Kitty dataset

huangxinCAS, 2019-03-25:
Hi, in author's example,  the version of stereo+imu  is  tested  on the EuRoc dataset. Can this version be used for kitty dataset? Will it work better than the stereo version?

shaozu, 2019-03-25:
Hi, @huangxinCAS. KITTI odometry dataset only contains IMU at 1HZ, so it's meaningless to do stereo+imu on it.

huangxinCAS, 2019-03-25:
Thanks for your reply, I ignored the frequency of imu before. I still have a question, I run the stereo version on kitty dataset, and there are two results: vio.txt and vio.csv. Are they equivalent?

shaozu, 2019-03-25:
Not exactly. you may refer [vio.txt](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/master/vins_estimator/src/KITTIOdomTest.cpp#L111) and [vio.csv](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/master/vins_estimator/src/utility/visualization.cpp#L148) for the meaning of these two files.

closed 2019-03-25 by shaozu

TonyPhh, 2021-11-05:
> Hi, @huangxinCAS. KITTI odometry dataset only contains IMU at 1HZ, so it's meaningless to do stereo+imu on it.

hi, I have a question, where can I find the IMU data for the KITTI odometry dataset.

RoyAmoyal, 2024-08-18:
> Hi, @huangxinCAS. KITTI odometry dataset only contains IMU at 1HZ, so it's meaningless to do stereo+imu on it.

Kitti actually has 10hz IMU. (synchronized with the camera).
Why can't I utilize the imu for more data using extended kalman filter for example?