HKUST-Aerial-Robotics/VINS-Fusion #31: Save and load map

Changliu52, 2019-04-07:
Thank you sharing the code.

I am wondering if saving and reloading the map is still available in this version, similar to vins-mono?

Thanks

mmp52, 2020-01-20:
hey @Changliu52 , you can save the map (pose graph and the features) by entering the letter s and pressing the enter at the terminal in which loop-fusion node is running. Making this closes the node and saves the posegraph. You can see the related code at [here](https://github.com/pjrambo/VINS-Fusion-gpu/blob/9ec9283d9d2348df05d99ed05bb8ecd8e4687efc/loop_fusion/src/pose_graph_node.cpp#L382)

luoxia-wen, 2021-04-27:
I entered the letter s and pressing the enter at the terminal "  rosrun vins vins_node ~/catkin_ws/src/VINS-Fusion/config/euroc/euroc_stereo_imu_config.yaml ", but nothing happend and no posegraph had been saved.

nirajreginald, 2021-04-29:
Did anyone solve this issue?

luoxia-wen, 2021-04-29:
I might find a way to solve this issue. It is like that we have to run the loop_fusion node(the "optional") to save the pose_graph and trajectory results. If we only want to get the trajectory result, we just run the loop_fusion(optional) node with other node and the results will be saved in the OUT_PUT_PATH. If we still want to get the pose_graph,  we have to enter the letter 's' and press the enter at the terminal in which loop-fusion node is running

HanDaSeul, 2023-08-31:
I had struggled with the issue the both vio_loop.csv and pose_graph.txt are saved 0 byte useless file, but it fixed with running 'vins_estimator', not 'vins_node'.

Sivan666666, 2024-01-15:
@HanDaSeul Hi, i have the same problem that the pose.graph.txt is saved 0 byte file. But my system tells vins_estimator not an executable name file. Can you tell the running procedure in specific code?

20221023lijia, 2024-08-06:
> I had struggled with the issue the both vio_loop.csv and pose_graph.txt are saved 0 byte useless file. That had been fixed by running 'vins_estimator', not running 'vins_node'.

hello!could you say it in details?what's your command?