HKUST-Aerial-Robotics/VINS-Fusion #35: roslaunch problem

metalsalmon, 2019-04-25:


metalsalmon, 2019-04-25:
When i start vins from roslaunch i can see only tracking images, no other topics are published.

<launch>	
  <include file="$(find vins)/launch/vins_rviz.launch"/>
  <node pkg="vins" type="vins_node" name="vins_node" args="~/catkin_ws/src/VINS-Fusion/config/euroc/euroc_stereo_imu_config.yaml" output="screen"/>
	
</launch>

this command works
rosrun vins vins_node ~/catkin_ws/src/VINS-Fusion/config/euroc/euroc_stereo_imu_config.yaml

Zhan-Jia-Rong, 2021-05-04:
I have this problem too!
I can't get rviz's visualization,when i use roslaunch to run "vins_node".
But it is normal to execute one by one.
Here is my roslaunch file:

```
<?xml version="1.0"?>
<launch>
    <include file="$(find vins)/launch/vins_rviz.launch">
	</include>
    <arg name="config_path" default = "/home/ccone/catkin_ws/src/VINS-Fusion/config/realsense_d455/realsense_mono_imu_config.yaml" />
    <node name="vins_fusion" pkg="vins" type="vins_node" output="screen">
        <param name="config_file" value="$(arg config_path)" />
    </node>
    <node pkg="rosbag" type="play" name="player" args="/home/jiarong/catkin_ws/vins+uwb/Date_4_21/cal3.bag"/>
</launch>
```

nlaby, 2024-03-18:
I made a launch file that works for me, here it is:

`<launch>

  <arg  name="bag_file" default="/home/dream/DATASETS/R3live/hkust_campus_03.bag"/>

  <arg name="publish_clock" default="--clock"/>

    
  <node pkg="image_transport" required="true" type="republish" name="compressed_fc" args="compressed in:=/camera/image_color  raw out:=/camera/image_color" />

  
  <node name="vins_estimator" required="true" pkg="vins" type="vins_node" output="screen" args="$(find vins)/../config/R3live/R3live_config.yaml"/>

  
  <node name="loop_fusion" pkg="loop_fusion" type="loop_fusion_node" output="screen" args="$(find vins)/../config/R3live/R3live_config.yaml"/>

    
  <node name="rvizvisualisation" respawn="true" pkg="rviz" type="rviz" output="log" args="-d $(find vins)/../config/vins_rviz_config.rviz" />

  

  <node required="false" pkg="rosbag" type="play" name="rosbag_player" args="$(arg publish_clock) $(arg bag_file) -r 1.0 " launch-prefix="bash -c 'sleep 1; $0 $@' "/>
  

</launch>`

nlaby, 2024-03-18:
This launch file works for me ;)

``` launch
<launch>

  <arg  name="bag_file" default="/home/dream/DATASETS/R3live/hkust_campus_03.bag"/>

  <arg name="publish_clock" default="--clock"/>

  <!-- Republish compressed image to raw image -->  
  <node pkg="image_transport" required="true" type="republish" name="compressed_fc" args="compressed in:=/camera/image_color  raw out:=/camera/image_color" />

  <!-- Run vins_node -->
  <node name="vins_estimator" required="true" pkg="vins" type="vins_node" output="screen" args="$(find vins)/../config/R3live/R3live_config.yaml"/>

  <!-- Run loop_fusion_node -->
  <node name="loop_fusion" pkg="loop_fusion" type="loop_fusion_node" output="screen" args="$(find vins)/../config/R3live/R3live_config.yaml"/>

    <!-- Launch RViz -->
  <node name="rvizvisualisation" respawn="true" pkg="rviz" type="rviz" output="log" args="-d $(find vins)/../config/vins_rviz_config.rviz" />

  <!-- Play rosbag -->

  <node required="false" pkg="rosbag" type="play" name="rosbag_player" args="$(arg publish_clock) $(arg bag_file) -r 1.0 " launch-prefix="bash -c 'sleep 1; $0 $@' "/>
  

</launch>

```