HKUST-Aerial-Robotics/VINS-Fusion #44: catkin make problem

yolingen23, 2019-05-18:
i have catkin make problem. 

`/home/gurturk/catkin_ws/src/VINS-Fusion/camera_models/src/chessboard/Chessboard.cc: In member function ‘void camodocal::Chessboard::generateQuads(std::vector<boost::shared_ptr<camodocal::ChessboardQuad> >&, cv::Mat&, int, int, bool)’:
/home/gurturk/catkin_ws/src/VINS-Fusion/camera_models/src/chessboard/Chessboard.cc:1175:50: error: ‘CV_RETR_CCOMP’ was not declared in this scope
     cv::findContours(image, contours, hierarchy, CV_RETR_CCOMP, CV_CHAIN_APPROX_SIMPLE);
                                                  ^~~~~~~~~~~~~
/home/gurturk/catkin_ws/src/VINS-Fusion/camera_models/src/chessboard/Chessboard.cc:1175:65: error: ‘CV_CHAIN_APPROX_SIMPLE’ was not declared in this scope
     cv::findContours(image, contours, hierarchy, CV_RETR_CCOMP, CV_CHAIN_APPROX_SIMPLE);
                                                                 ^~~~~~~~~~~~~~~~~~~~~~
VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/build.make:110: recipe for target 'VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/src/calib/CameraCalibration.cc.o' failed
make[2]: *** [VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/src/calib/CameraCalibration.cc.o] Error 1
/home/gurturk/catkin_ws/src/VINS-Fusion/camera_models/src/chessboard/Chessboard.cc:1241:27: error: ‘CV_CALIB_CB_FILTER_QUADS’ was not declared in this scope
             if (!(flags & CV_CALIB_CB_FILTER_QUADS) ||
                           ^~~~~~~~~~~~~~~~~~~~~~~~
VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/build.make:86: recipe for target 'VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/calib/CameraCalibration.cc.o' failed
make[2]: *** [VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/calib/CameraCalibration.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/gurturk/catkin_ws/src/VINS-Fusion/camera_models/src/chessboard/Chessboard.cc: In member function ‘bool camodocal::Chessboard::checkChessboard(const cv::Mat&, cv::Size) const’:
/home/gurturk/catkin_ws/src/VINS-Fusion/camera_models/src/chessboard/Chessboard.cc:1586:72: error: ‘CV_THRESH_BINARY’ was not declared in this scope
         cv::threshold(white, thresh, threshLevel + blackWhiteGap, 255, CV_THRESH_BINARY);
                                                                        ^~~~~~~~~~~~~~~~
/home/gurturk/catkin_ws/src/VINS-Fusion/camera_models/src/chessboard/Chessboard.cc:1592:55: error: ‘CV_RETR_CCOMP’ was not declared in this scope
         cv::findContours(thresh, contours, hierarchy, CV_RETR_CCOMP, CV_CHAIN_APPROX_SIMPLE);
                                                       ^~~~~~~~~~~~~
/home/gurturk/catkin_ws/src/VINS-Fusion/camera_models/src/chessboard/Chessboard.cc:1592:70: error: ‘CV_CHAIN_APPROX_SIMPLE’ was not declared in this scope
         cv::findContours(thresh, contours, hierarchy, CV_RETR_CCOMP, CV_CHAIN_APPROX_SIMPLE);
                                                                      ^~~~~~~~~~~~~~~~~~~~~~
/home/gurturk/catkin_ws/src/VINS-Fusion/camera_models/src/chessboard/Chessboard.cc:1597:56: error: ‘CV_THRESH_BINARY_INV’ was not declared in this scope
         cv::threshold(black, thresh, threshLevel, 255, CV_THRESH_BINARY_INV);
                                                        ^~~~~~~~~~~~~~~~~~~~
VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/build.make:62: recipe for target 'VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/src/intrinsic_calib.cc.o' failed
make[2]: *** [VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/src/intrinsic_calib.cc.o] Error 1
CMakeFiles/Makefile2:676: recipe for target 'VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/all' failed
make[1]: *** [VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/build.make:62: recipe for target 'VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/chessboard/Chessboard.cc.o' failed
make[2]: *** [VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/chessboard/Chessboard.cc.o] Error 1
CMakeFiles/Makefile2:447: recipe for target 'VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/all' failed
make[1]: *** [VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`

superhector2000, 2019-10-16:
Did you solve this issue? I am having the same problem!

IaroslavS, 2020-04-30:
I had the same problem. I uninstalled opencv 4.3.0 and installed opencv 3.3.1. Now it works.

jagauthier, 2020-05-11:
> I had the same problem. I uninstalled opencv 4.3.0 and installed opencv 3.3.1. Now it works.

I don't see OpenCV on the list of prerequisites.

natetoombs, 2021-01-14:
If you are on Ubuntu 20, [this pull request](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/pull/364/commits/5f1561aa573016dac778005a7274352515014946) for VINS-mono fixed the problem for me!

JayLuoShui, 2022-03-11:
楼上的方法确实顶！