HKUST-Aerial-Robotics/VINS-Fusion #48: I don't quite understand the meaning of the following parameters. Can you explain?

HKVINS89757, 2019-06-11:
body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ 4.2812441490024389e-03, -9.9997001507473682e-01,
       -6.4528985710044385e-03, 5.2583356071589790e-05,
       9.9996900935734523e-01, 4.2384270612576547e-03,
       6.6344601088757426e-03, -4.2174706544162562e-02,
       -6.6069110351583190e-03, -6.4811023350536514e-03,
       9.9995717110239080e-01, 1.9238715201769417e-02, 0., 0., 0., 1. ]

body_T_cam1: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ 1.6525177162010074e-03, -9.9998509853672479e-01,
       -5.2030654132755290e-03, -2.9251875934051923e-03,
       9.9994935393851081e-01, 1.6007594247438028e-03,
       9.9361525360214843e-03, 7.4379061731550719e-02,
       -9.9276756168115085e-03, -5.2192215666013279e-03,
       9.9993709851324453e-01, 1.9302672329335892e-02, 0., 0., 0., 1. ]

HKVINS89757, 2019-06-11:
哪位大神解释一下呀 
这俩矩阵什么意思呀？
我没有用imu 用的普通的摄像头  

我不太明白这些错误 我的电脑上给出的错误是：
^[^AWARNING: Logging before InitGoogleLogging() is written to STDERR
W0611 20:00:38.735064 27020 residual_block.cc:130] 

Error in evaluating the ResidualBlock.

There are two possible reasons. Either the CostFunction did not evaluate and fill all    
residual and jacobians that were requested or there was a non-finite value (nan/infinite)
generated during the or jacobian computation. 

Residual Block size: 5 parameter blocks x 2 residuals

For each parameter block, the value of the parameters are printed in the first column   
and the value of the jacobian under the corresponding residual. If a ParameterBlock was 
held constant then the corresponding jacobian is printed as 'Not Computed'. If an entry 
of the Jacobian/residual array was requested but was not written to by user code, it is 
indicated by 'Uninitialized'. This is an error. Residuals or Jacobian values evaluating 
to Inf or NaN is also an error.

HKVINS89757, 2019-06-11:
who can explained it?
What do these two matrix mean?
I  using Stereo camera without imu

The error message that appears on my computer is:
WARNING: Logging before InitGoogleLogging() is written to STDERR
W0611 20:00:38.735064 27020 residual_block.cc:130]

Error in evaluating the ResidualBlock.

There are two possible reasons. Either the CostFunction did not evaluate and fill all
residual and jacobians that were requested or there was a non-finite value (nan/infinite)
generated during the or jacobian computation.

Residual Block size: 5 parameter blocks x 2 residuals

For each parameter block, the value of the parameters are printed in the first column
and the value of the jacobian under the corresponding residual. If a ParameterBlock was
held constant then the corresponding jacobian is printed as 'Not Computed'. If an entry
of the Jacobian/residual array was requested but was not written to by user code, it is
indicated by 'Uninitialized'. This is an error. Residuals or Jacobian values evaluating
to Inf or NaN is also an error.

mpkuse, 2019-06-17:
body_T_cam0 means when you are sitting on body (imu) the pose (position and orientation) of camera0 (left camera). 

So, body_T_cam0 is the pose of the cam0 as observed from body frame aka imu. 

Hope this helps.

HKVINS89757, 2019-06-17:
> body_T_cam0 means when you are sitting on body (imu) the pose (position and orientation) of camera0 (left camera).
> 
> So, body_T_cam0 is the pose of the cam0 as observed from body frame aka imu.
> 
> Hope this helps.
ths my bro

closed 2019-06-23 by pjrambo

haithink, 2019-12-09:
> who can explained it?
> What do these two matrix mean?
> I using Stereo camera without imu
>
> The error message that appears on my computer is:
> WARNING: Logging before InitGoogleLogging() is written to [...]

same problem if running without imu data

takmin, 2020-04-02:
I think body_T_cam0 is NOT camera 0 pose in body (IMU) coordinates.
I read source code and body_T_cam0 seems body pose in camera0 coordinates, which mean point in body (IMU) homogeneous coordinates p is transformed to camera0 coordinates q by:

q = body_T_cam0 * p

xdtl, 2020-05-13:
> I think body_T_cam0 is NOT camera 0 pose in body (IMU) coordinates.
> I read source code and body_T_cam0 seems body pose in camera0 coordinates, which mean point in body (IMU) homogeneous coordinates p is transformed to camera0 coordinates q by:
> 
> q = body_T_cam0 * p

Could somebody please confirm if this equation is true? If so, then this is the transformation from body coordinates to camera coordinates. Thanks!

shaozu, 2020-05-14:
> body_T_cam0 means when you are sitting on body (imu) the pose (position and orientation) of camera0 (left camera).
> 
> So, body_T_cam0 is the pose of the cam0 as observed from body frame aka imu.
> 
> Hope this helps.

This answer is correct.

xdtl, 2020-05-14:
> > body_T_cam0 means when you are sitting on body (imu) the pose (position and orientation) of camera0 (left camera).
> > So, body_T_cam0 is the pose of the cam0 as observed from body frame aka imu.
> > Hope this helps.
> 
> This answer is correct.

Thanks for your reply! I mean if the relation can be described as an equation such as q = body_T_cam0 * p, it would be easier to understand. I feel it is still a bit confusing by descriptions like "pose of the camera", when we are talking about a 4x4 transformation matrix.

takmin, 2020-05-21:
> > body_T_cam0 means when you are sitting on body (imu) the pose (position and orientation) of camera0 (left camera).
> > So, body_T_cam0 is the pose of the cam0 as observed from body frame aka imu.
> > Hope this helps.
> 
> This answer is correct.

Do you mean the equation below?:

p = body_T_cam0 * q

p: point in IMU coordinates
q: point in camera coordinates

shaozu, 2020-05-21:
> Do you mean the equation below?:
>
> p = body_T_cam0 * q
>
> p: point in IMU coordinates
> q: point in camera coordinates

Yes

xdtl, 2020-05-21:
@takmin Thanks for your followup! I notice the equation you posted on April 1 is exactly opposite to what you posted yesterday.
@shaozu Thank you for the answer! So the equation is: p_imu = body_T_cam0 * p_camera.