HKUST-Aerial-Robotics/VINS-Fusion #53: A large drifting with Realsense D435i

quxichao, 2019-06-20:
Hello, I recently used D435i to collect data for performance testing.

the rs_camera.launch is
`<launch>
  <arg name="serial_no"           default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="true"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="250"/> 
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  
  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"           default="true"/>
  <arg name="align_depth"           default="true"/>

  <arg name="filters"               default=""/>
  <arg name="clip_distance"         default="-2"/>
  <arg name="linear_accel_cov"      default="0.01"/>
  <arg name="initial_reset"         default="false"/>
  <arg name="unite_imu_method"      default="linear_interpolation"/>
  <arg name="topic_odom_in"         default="odom_in"/>
  <arg name="calib_odom_file"       default=""/>
  <arg name="publish_odom_tf"       default="true"/>
  <arg name="hold_back_imu_for_frames"      default="true"/>
  
  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
      <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye"           value="$(arg enable_fisheye)"/>

      <arg name="depth_width"              value="$(arg depth_width)"/>
      <arg name="depth_height"             value="$(arg depth_height)"/>
      <arg name="enable_depth"             value="$(arg enable_depth)"/>

      <arg name="color_width"              value="$(arg color_width)"/>
      <arg name="color_height"             value="$(arg color_height)"/>
      <arg name="enable_color"             value="$(arg enable_color)"/>

      <arg name="infra_width"              value="$(arg infra_width)"/>
      <arg name="infra_height"             value="$(arg infra_height)"/>
      <arg name="enable_infra1"            value="$(arg enable_infra1)"/>
      <arg name="enable_infra2"            value="$(arg enable_infra2)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>
 
      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>

     </include>
  </group>
</launch>`

the realsense_stereo_imu_config.yaml is
`%YAML:1.0

#common parameters
#support: 1 imu 1 cam; 1 imu 2 cam: 2 cam; 
imu: 1         
num_of_cam: 2  

imu_topic: "/camera/imu"
image0_topic: "/camera/infra1/image_rect_raw"
image1_topic: "/camera/infra2/image_rect_raw"
output_path: "/home/quxichao/catkin-ws/output/"

cam0_calib: "left.yaml"
cam1_calib: "right.yaml"
image_width: 640
image_height: 480
   

# Extrinsic parameter between IMU and Camera.
estimate_extrinsic: 1   # 0  Have an accurate extrinsic parameters. We will trust the following imu^R_cam, imu^T_cam, don't change it.
                        # 1  Have an initial guess about extrinsic parameters. We will optimize around your initial guess.

body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ -5.7586305857286746e-03, -4.0463318787729019e-03,
       9.9997523237933461e-01, 2.0329267950355900e-02,
       -9.9998287214160420e-01, -1.0224590553211677e-03,
       -5.7628118925283633e-03, 7.9325209639615653e-03,
       1.0457519809151661e-03, -9.9999129084997906e-01,
       -4.0403746097850135e-03, 2.8559824645148020e-03, 0., 0., 0., 1. ]

body_T_cam1: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ -1.0021770212322867e-03, 3.6313480322730518e-04,
       9.9999943188700535e-01, 1.5285779565991807e-02,
       -9.9999216342926500e-01, -3.8303422615924010e-03,
       -1.0007788055728661e-03, -5.2435791444330505e-02,
       3.8299766679101843e-03, -9.9999259827824449e-01,
       3.6697063849344680e-04, 8.6931302450199057e-03, 0., 0., 0., 1. ]

#Multiple thread support
multiple_thread: 1

#feature traker paprameters
max_cnt: 150            # max feature number in feature tracking
min_dist: 30            # min distance between two features 
freq: 10                # frequence (Hz) of publish tracking result. At least 10Hz for good estimation. If set 0, the frequence will be same as raw image 
F_threshold: 1.0        # ransac threshold (pixel)
show_track: 0           # publish tracking image as topic
flow_back: 1            # perform forward and backward optical flow to improve feature tracking accuracy

#optimization parameters
max_solver_time: 0.04  # max solver itration time (ms), to guarantee real time
max_num_iterations: 8   # max solver itrations, to guarantee real time
keyframe_parallax: 10.0 # keyframe selection threshold (pixel)

#imu parameters       The more accurate parameters you provide, the better performance
acc_n: 0.1          # accelerometer measurement noise standard deviation. #0.2   0.04
gyr_n: 0.01         # gyroscope measurement noise standard deviation.     #0.05  0.004
acc_w: 0.001         # accelerometer bias random work noise standard deviation.  #0.002
gyr_w: 0.0001       # gyroscope bias random work noise standard deviation.     #4.0e-5
g_norm: 9.805         # gravity magnitude

#unsynchronization parameters
estimate_td: 1                      # online estimate time offset between camera and imu
td: 0.00                             # initial value of time offset. unit: s. readed image clock + td = real image clock (IMU clock)

#loop closure parameters
load_previous_pose_graph: 0        # load and reuse previous pose graph; load from 'pose_graph_save_path'
pose_graph_save_path: "/home/quxichao/catkin-ws/output/pose_graph/" # save and load path
save_image: 0                   # save image in pose graph for visualization prupose; you can close this function by setting 0 `

When I run the code，the track_image is none,and the location is drifting 
[image]
[image]

HKVINS89757, 2019-06-23:
感觉遇到的问题跟你一样

pjrambo, 2019-06-23:
@quxichao  Please check the firmware version you used. The confing is for SDK2.19. The version of Realsense SDK higher than 2.19 exists some issue with its IMU. You can directly using the config file with SDK 2.19.

HKVINS89757, 2019-06-23:
> @quxichao  Please check the firmware version you used. The confing is for SDK2.19. The version of Realsense SDK higher than 2.19 exists some issue with its IMU. You can directly using the config file with SDK 2.19.

If using dual-camera only without IMU。  Will this problem arise?

pjrambo, 2019-06-23:
> > @quxichao  Please check the firmware version you used. The confing is for SDK2.19. The version of Realsense SDK higher than 2.19 exists some issue with its IMU. You can directly using the config file with SDK 2.19.
> 
> If using dual-camera only without IMU。 Will this problem arise?

You can try it yourself. If your usage is right, this problem should not happen.

HKVINS89757, 2019-06-23:
> @quxichao  Please check the firmware version you used. The confing is for SDK2.19. The version of Realsense SDK higher than 2.19 exists some issue with its IMU. You can directly using the config [...]

But on my computer， VINS cant provide satisfactory result.   
The path is very different from the real scene

HKVINS89757, 2019-06-23:
> @quxichao  Please check the firmware version you used. The confing is for SDK2.19. The version of Realsense SDK higher than 2.19 exists some issue with its IMU. You can directly using the config [...]

I using dual-camera and Calibrate them by kalibr.   

My cameras are 120degree angle.

HKVINS89757, 2019-06-23:
> @quxichao  Please check the firmware version you used. The confing is for SDK2.19. The version of Realsense SDK higher than 2.19 exists some issue with its IMU. You can directly using the config [...]

I tried to calibrate my cameras by different types. such as mei & pinhole

HKVINS89757, 2019-06-23:
> @quxichao  Please check the firmware version you used. The confing is for SDK2.19. The version of Realsense SDK higher than 2.19 exists some issue with its IMU. You can directly using the config [...]

Config file as follows, can you give me some tips and advice?:
[image]
[image]

shaoxiang, 2019-07-12:
@quxichao @HKVINS89757 The same problem, has anyone solved it?  I tried switching SDK to version 2.18.1, but it was worse.

HKVINS89757, 2019-07-12:
> @quxichao @HKVINS89757 The same problem, has anyone solved it?  I tried switching SDK to version 2.18.1, but it was worse.

说中文

AilsonAir, 2019-07-12:
I was also running vins-fusion with d435i. The tracking works fine with SDK 2.18. But I also can't have any tracked_image shown on the left. Any idea?

SainaRez, 2019-08-15:
@pjrambo Which Intel firmware version are you guys using on D435i?

cichard, 2019-09-20:
Hello 👋🏻 ! Many issues here .... 
First consider switching on the sdk 2.25 and ros corresponding 2.2.8 because they fix many problem on the imu part. Then use the last production firmware 11.2.6.250

-Then do not use their linear interpolation implementation : because as you will be able to see it is buggy ! It is a mixed interpolation between fast and slow stream so as you will be able to see time stamp are messed up ! 
- Then consider using imu_tools to compute Allan parameters of your device : you will have higher value than those provided by the config file in the github repository of vins fusion : impressive 
-Then I will recommend you to use global time flag in the ros node of RealSense to get time of capture instead of time of reception message and better do not use ros node provided by realsense but one called rs4se which makes a good implementation for visual inertial system 
- use Kalibr to make the calibration between cameras and camera/imu 
- take into account that vins convention seems to wait for a z up body frame in the camera to body projection matrix 
- vins is waiting for imu data that is corrected for misalignment and bias, the realsense documentation do state that the operation is performed in the firmware after performing the es-imu-calibration script. But as I could have seen, there is a bug in the correction, so you will have to disable this correction by setting the option “enable motion correction “ and implement it by yourself in the realsense node 
! 
Hopefully it can help anyone

zgxsin, 2019-11-05:
> You can try it yourself. If your usage is right, this problem should not happen.

Hi @quxichao Thanks for your answer here. Here is the timestamp of the images and imu measurements from realsense D435i:
[image]
By visualization, the IMU and image seem to be synced. But VINS-Fusion with stereo_imu configuration just drifted aways largely. I also ran VINS-Fusion with only stereo images from the realsense, it worked.
Do you have any idea about this issue?
Thank you!

mhc994, 2019-12-15:
I have just run vins successfully with a realsense D435i camera. Stereo + IMU config is fine.
The success relies heavily on versions.

As a suggestion, you may try:
ubuntu 18.04 + ROS melodic (I think 16.04+ROS Kinetic shoud also work)
Linux kernel 4.15.0-72-generic (5.0 didn't work for me, be sure to reinstall librealsense2-dkms after changing kernel)
librealsense2 2.19.0-0\~realsense0.615 
librealsense2-dev 2.19.0-0\~realsense0.615 (2.19.2 didn't work for me) 
librealsense2-dkms 1.3.10-0ubuntu1
realsense2_camera 2.2.3 (just git checkout 2.2.3 and catkin_make)
a realsense launch file at [https://github.com/mhc994/VINS-Fusion/blob/wz_fly/config/realsense_d435i/vins_infra.launch](url) (I disabled useless depth and color topics)

good luck
[image]

ceccocats, 2020-08-20:
Hi @mhc994 
Im trying d435i and succefully make VINS working with 1 imu and 1 camera but the stereo is not working.
Im using SDK 2.36 and rs4se to grab the data.
The calibration made with kalibr is good with very low errors, but still I have awful result when using stereo.
I think that one issue can be the calibration of the IMU that is different from yours, my calib is the right one.
As you can see the IMU position is rotated:
[image]
Do you experienced the same issue with the different versions you used?

Im going to try your setup but I have problems to start my pc with linux kernel 4.15, can you tell me what type of problem you had with kernel 5?

thanks

dyb678, 2024-11-02:
> @HKVINS89757 同样的问题，有人解决过吗？我尝试将 SDK 切换到版本 2.18.1，但情况更糟。 我换了SDK版本后， 还是没有解决这个问题 [image]

Hello~do you solve this problem?Actually , I find that the mininum version for D435i is 2.36.0. How can I install 2.18.0?
[image]