HKUST-Aerial-Robotics/VINS-Fusion #55: Intrinsic and extrinsic calibration

SainaRez, 2019-06-27:
How did you get the extrinsic calibration parameters for realsense D435i and imu and the imu intrinsic parameters? Did you use a specific package?

HKVINS89757, 2019-06-28:
kalibr

closed 2019-07-17 by SainaRez

zgxsin, 2019-11-06:
Hi @HKVINS89757 , can i get the extrinsic transformation between camera and IMU for realsense D435i from the tf tree by launching `realsense2_camera` package?

Here is what i did:

1.  Echo image topic and check its frame_id
[image]

1. Echo imu topic and check its frame_id
[image]

1. Use tf_echo to get the transformation between from image to  imu.
[image]

I compare this result to [here](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/master/config/realsense_d435i/realsense_stereo_imu_config.yaml). They are not consistent at all. Any idea?

shirbarzel, 2020-07-20:
Hi @zgxsin I saw you mentioned another issue but still I don't understand how to update my realsense_stereo_imu_config.yaml according to my camera?
Can you please elaborate?