HKUST-Aerial-Robotics/VINS-Fusion #56: DDDDrift！！！！ binocular stereo camera test reslut！Why so large Drift？？

HKVINS89757, 2019-07-04:
Hi:
I tested VINS-FUSION using  binocular stereo camera without IMU.
BBBUT, I got a large Drift.
Actually，I moved camera only just a few centimeters
But，VINS-FUSION program calculation result is More than 800km？？！！
[image]

I checked the rostopic， it says as follows：
[image]

HKVINS89757, 2019-07-04:
here is my extrinsic file：
[image]

AilsonAir, 2019-07-23:
How do you get the camera position? Is it an absolute or relative position?
@HKVINS89757

Ken-Thorburn, 2019-07-30:
Were you able to figure this out? I have a similar problem.

TabooTY, 2019-08-22:
The same situation occurs to me when I use my own dataset. I also got a large drift. I guess it may be the features' problem.

whyscience, 2019-08-22:
[image]
I got one like this

zgxsin, 2019-10-28:
I also have the same problem here. 
[image].
But sometimes it works well. I think it is very sensitive to initialization.