HKUST-Aerial-Robotics/VINS-Fusion #66: vins_node cpu usage is almost 300%

zhouchengming1, 2019-08-14:
Hi all,
I run vins_node on Nvidia Xavier board, the cpu usage of one vins_node is almost 300%.
After I profile using perf tools, I found：
1. Estimator::inputImage  use too much cv::parallel_for_ and trigger too much work on TBB thread.
2. Estimator::processMeasurements use MarginalizationInfo::marginalize, and this create too much threads to do work...

I don't know much about vins theory...  so anyone can help me with this performance problem?
Thanks.

zhouchengming1, 2019-08-14:
Hi, Any ideas?

append attachment about MarginalizationInfo::marginalize 
[image]

zhouchengming1, 2019-08-15:
[image]

And the strange thing is that although I changed to this no-pthread in-place calculation version.
the cpu usage does not go down, but the percentage of ceres::Solve go up...

[image]

abdul-mannan-khan, 2024-08-28:
I am facing the same problem. Really looking for help. @zhouchengming1 Could you get any clue? Thank you for your answer.