HKUST-Aerial-Robotics/VINS-Fusion #73: Why do you remove the relocalization part from the vins_estimator

skyertian, 2019-09-15:
In VINS-MONO, in estimator_node you subscribe from topic "/pose_graph/match_points" messages published by the pose_graph_node and add relocalization residual in the problem. ( in estimator.cpp, there is "problem.AddParameterBlock(relo_Pose, SIZE_POSE, local_parameterization);" ).

However, in VIN-FUSION, you removed the relocalization part in vins_estimator and the problem doesn't contain relocalization residual any more. Why is that ?

mmp52, 2020-01-16:
@skyertian have you find out why or did you try with re-inserting that part?

Shame-fight, 2020-05-27:
The most important part was removed,so pity

ymd-stella, 2020-09-28:
You can use https://github.com/mpkuse/cerebro.