HKUST-Aerial-Robotics/VINS-Fusion #74: How to modify (from code) the information matrix (covariance matrix) of   Visual Measurement Residual

weisongwen, 2019-09-16:
Hi @shaojie @shaozu , thanks for sharing the great work.  Thanks a lot. 
I am reading your work and testing your code using the real data collected from road. We find that sometimes the uncertainty (corresponding to the information matrix in factor graph) of  _Visual Measurement Residual_ should be larger in challenging environments with numerous dynamic objects (vehicles). 
[image]

if we already know the uncertainty of the _Visual Measurement Residual_.  How can we modify the information in the vins-fusion code or how it is estimated?

Thanks again for sharing the your great work,

Best,
Weisong,

hridaybavle, 2020-04-22:
Hi @weisongwen did you manage to change the information matrix for the visual features?

WFram, 2023-01-25:
It's already fixed. See the following code in `estimator.cpp` :

```
ProjectionTwoFrameOneCamFactor::sqrt_info = FOCAL_LENGTH / 1.5 * Matrix2d::Identity();
ProjectionTwoFrameTwoCamFactor::sqrt_info = FOCAL_LENGTH / 1.5 * Matrix2d::Identity();
ProjectionOneFrameTwoCamFactor::sqrt_info = FOCAL_LENGTH / 1.5 * Matrix2d::Identity();
```