HKUST-Aerial-Robotics/VINS-Fusion #76: FOCAL_LENGTH problem!

baaixw, 2019-09-20:
Dear @shaojie @shaozu , thanks for your sharing the great work.
I read your code and test it using the real data collected from road. 
Now, i meet a problem that : 
1. ProjectionTwoFrameOneCamFactor::sqrt_info = FOCAL_LENGTH / 1.5 * Matrix2d::Identity();
How to get the exact FOCAL_LENGTH?  

##   If anyone knows the method, please contact me .
Thank you so much.

baaixw, 2019-09-20:
i mean if i use my own camera, how do i change the value?

shaozu, 2019-10-22:
@baaixw It's just a virtual parameter, so you don't need to change it anyway.

baaixw, 2019-10-24:
> @baaixw It's just a virtual parameter, so you don't need to change it anyway.

thanks for your kindly reply.