HKUST-Aerial-Robotics/VINS-Fusion #77: Why did the rotation matrix change too much when the vio was fused with GPS?

HpyZhai, 2019-10-08:
I tested the global fusion node with the KITTI data. I found that the car model's pose in the rviz was abnormal. Sometimes the car's nose was not forward but up or towards other directions. I printed the veriable  " WGPS_T_WVIO ",  and found the rotation matrix always changed too much as the ego-car in the real world made a turn, however the translation vector didn't change too much. 
     Did it cause by the lack of the orientation constraint in the fusing processing?  Dose the fused result get better with  another sensor-----magnetometer?

weisongwen, 2019-10-17:
Hi @HpyZhai , have you fixed this problem? I see the same issue.

zxh2015, 2020-04-14:
I can't understand why we should compute the WGPS_T_WVIO matrix so many times. And I think the transformation matrix of GPS and VIO should  be fixed when we received first gps frame.Can someone give me an answer?

HpyZhai, 2020-04-16:
> I can't understand why we should compute the WGPS_T_WVIO matrix so many times. And I think the transformation matrix of GPS and VIO should be fixed when we received first gps frame.Can someone give me an answer?

我理解的是 从body角度看，world是处于漂移的状态 (和从world角度看， body的位姿存在漂移一样)。用每时每刻的值来计算WGPS_T_WVIO，它是变化的。WGPS_T_WVIO = w_GPS_T_body*body_T_WVIO;

HpyZhai, 2020-04-16:
> Hi @HpyZhai , have you fixed this problem? I see the same issue.

没有解决，我觉得可能是忽略了相机和gps之间的距离引起的。相机和 gps应该是两个body，论文代码把他们当作同一个body。如果是无人机，由于空间有限，gps和相机的安装距离很近，可以忽略，如果是车载，这个距离就不能忽略(参考kitti 传感器的安装)，会对结果造成影响。

zxh2015, 2020-04-17:
> > Hi @HpyZhai , have you fixed this problem? I see the same issue.
> 
> 没有解决，我觉得可能是忽略了相机和gps之间的距离引起的。相机和 gps应该是两个body，论文代码把他们当作同一个body。如果是无人机，由于空间有限，gps和相机的安装距离很近，可以忽略，如果是车载，这个距离就不能忽略(参考kitti 传感器的安装)，会对结果造成影响。

是的,我也发现了这个问题,当我不使用kitti data的时候,由于gps与imu和camera安装位置很不同,这里的问题更加明显