HKUST-Aerial-Robotics/VINS-Fusion #79: Running with my own device(Realsense d435 + Auvidea J120's mpu9250) drift.

jackyfriend, 2019-10-15:
@qintonguav @shaozu @pjrambo 
Hello, I run vins-fusion on Jetson TX2 with camera Realsense d435 and an IMU of mpu9250 on the  J120 board, but encounter very large drift, I have no ideas of how to do now. Should I calibrate the camera and imu according to this websit [,](https://github.com/ethz-asl/kalibr) and not using estimate_extrinsic: 2? Please give me some advise. Thank you so much!

`rosrun vins vins_node ~/work/catkin_fusion/src/VINS-Fusion/config/d435_mpu9250/d435_mono_imu_config.yaml `
[image]

The d435_mono_imu_config.yaml is as follow:
[image]

IMU is using rtimulib:
[image]

And the drifting image is:
[image]

LuciousZhang, 2019-10-29:
According to my experience, you'd better calibrate the device yourself (I use imu_utils and kalibr). 
Not using `estimate_extrinsic` should input 0 instead of 2.

cherishyourself, 2023-05-16:
请问 解决了吗

cherishyourself, 2023-05-16:
Excuse me, have you solved it