HKUST-Aerial-Robotics/VINS-Fusion #82: Obvious position error when run car demo

ysong-aceinna, 2019-11-05:
Hi, @qintonguav @shaozu @pjrambo, thank you for your great work!

When I run the car demonstration, I find **Mono+IMU** with a big error than Stereo+IMU version  when disable loop closure. 

As below picture, the end position is [1.525746 -0.092251 0.107199], assume the start and end point with the same position, so the error is about 1.53m.

[image]

I meet same situation when I estimate mynteye S1030 camera with a built-in IMU. As below picture, total path length is about 130m and get 2 m error. Currently, I want to improve the accuracy, any suggestion will be appreciated!

[image]

Besides, could you please share which are the best camera and IMU you have estimated to run VINS? Thank you very much!

tuandle, 2019-11-05:
I think the drifting is inevitable when using mono+imu. Have you tried to turn on the loop closure?

ysong-aceinna, 2019-11-05:
Hi @tuandle, 

Yes, drift is inevitable, but I hope to improve the mono+imu accuracy for above 2 cases which obviously worse than paper claimed.

Loop-closure should be useful and can reduce drift significantly for above 2 cases. But it couldn't come in handy in some situations, such as limited hardware resource (CPU, memory) and unclosed path, so reduce drift is important.

HpyZhai, 2019-11-07:
Set the estimate_extrinsic true in the config file. The result might be better. using my own dataset, it worked . Although the estrinsic params optimized were very different from the calibrated ones, the vio trajectory was better.

ysong-aceinna, 2019-11-08:
Thank you @HpyZhai, 
I knew set the "estimate_extrinsic = 1" could make trajectory better sometimes, but not always better. As description of "estimate_extrinsic" option, we can set it to "1" when just have an initial guess about extrinsic parameters, so I trend to get a accuracy calibrate parameters and set it to "0".
Share you below picture which got by Mono+IMU and "estimate_extrinsic = 1",  a bit worse than above picture which set  "estimate_extrinsic = 0".

[image]