HKUST-Aerial-Robotics/VINS-Fusion #87: Wrong gps path

Andrea485, 2019-11-19:
The white path is the fundamental truth and the blue path is global fusion.
The global merger path is wrong.
The problem is the conversion from the global map (latitude, longitude and altitude) to local (xyz), but the code is ok. Are there any Geographic lib parameters that need to be set?
[image]

How can I solve this problem?

Michael-Liu516, 2019-12-03:
Did you solve this issue now? I'm confusing about this now. Could you please give some advices? Thanks

zxh2015, 2020-04-17:
Did you solve this issue now?

Sadegh-Kalami, 2023-08-27:
could pls help me how did u add Ground truth into RVIZ?
@andelby @Michael-Liu516