HKUST-Aerial-Robotics/VINS-Fusion #89: Assertion `px != 0' failed. Aborted (core dumped)

tekrajchhetri, 2019-11-20:
I am getting core dump issue even with the provided default configuration and also with my own. I ran MH_01 euroc dataset with imu. I also followed the suggestions from issue  #36 for the fix i.e. installing stable version of ceras and eigen.

[image]

Thank you.

jerrykingnice, 2019-11-25:
Hi
you need to notice,the cmd you type.
Instead of  type rosrun  vins vins_node euroc_mono_imu_config.yaml ,
you should give rosrun  vins vins_node **catkin_ws/src/VINS-Fusion/config/locate/euroc_mono_imu_config.yaml**

tekrajchhetri, 2019-11-25:
@jerrykingnice thanks for your response.  I was in the directory where the launch files are present so I just gave the file name. Anyway when i run following what you said, I got the following.

[image]

tekrajchhetri, 2019-11-30:
but now I am getting a gyroscope bias error. The calibration was performed using kalibr.

haithink, 2019-12-05:
I also encounttered this problem. But in some cases, there was not this problem And the only difference is the version of some thirdparty libraries such as boost or opencv.

tekrajchhetri, 2019-12-05:
@haithink Could you provide information on the version of the libraries you used? This is a little strange. Isn't it that as long as the method that it is currently using is available then it should work otherwise will everyone will be stick to use the same version?

haithink, 2019-12-06:
@tekrajchhetri  When I use boost1.71 , then no problem;  In boost 1.54 case, then assert failed. I have googled this problem, many people said that it occured because of using shared pointers without initialization.[https://stackoverflow.com/questions/3541179/shared-ptr-assertion-px-0-failed](refer this)

LouisValery, 2020-02-26:
I had this error. In my case it mas clearly  a problem linked to the parameters.
In particular the two camera files where not correctly set (in the euroc_mono_imu_config.yaml):
cam0_calib: "cam0_pinhole.yaml"
cam1_calib: "cam1_pinhole.yaml"

make sur that the names of these files are correct and that thes files are in the good folder.

tekrajchhetri, 2020-07-22:
closing this issue as i am no longer working with it.

closed 2020-07-22 by tekrajchhetri

trunc8, 2021-09-04:
> @jerrykingnice thanks for your response. I was in the directory where the launch files are present so I just gave the file name.

For the benefit of anyone else that stumbles on this issue, giving just the file name _will_ create this error. From the code in `parameters.cpp`:
```cpp
    int pn = config_file.find_last_of('/');
    std::string configPath = config_file.substr(0, pn);
```
So you require at least one directory name in your terminal argument. If you input just the filename, `pn=-1` and `configPath` will take the entire `config_file` string. This gives nonsensical file paths later, as is evident in the image of the first post in this thread-
`reading parameter of camera euroc_stereo_imu_config.yaml/cam0_mei.yaml`