HKUST-Aerial-Robotics/VINS-Fusion #90: waite for imu ...

cugcsu, 2019-11-22:
admi@admi-virtual-machine:~$ rosrun vins vins_node '/home/admi/VINS_fusion/src/config/realsense_d435i/realsense_stereo_imu_config.yaml' 
[ INFO] [1574410295.978970904]: init begins
config_file: /home/admi/VINS_fusion/src/config/realsense_d435i/realsense_stereo_imu_config.yaml
USE_IMU: 1
IMU_TOPIC: /camera/imu
result path /home/dji/output//vio.csv
[ WARN] [1574410295.983962786]:  Optimize extrinsic param around initial guess!
camera number 2
[ INFO] [1574410295.993746987]: Unsynchronized sensors, online estimate time offset, initial td: 0
[ INFO] [1574410295.993827005]: ROW: 480 COL: 640 
 exitrinsic cam 0
-0.00575863 -0.00404633    0.999975
  -0.999983 -0.00102246 -0.00576281
 0.00104575   -0.999991 -0.00404037
 0.0203293 0.00793252 0.00285598
 exitrinsic cam 1
-0.00100218 0.000363135    0.999999
  -0.999992 -0.00383034 -0.00100078
 0.00382998   -0.999993 0.000366971
 0.0152858 -0.0524358 0.00869313
set g     0     0 9.805
[ INFO] [1574410295.995070704]: reading paramerter of camera /home/admi/VINS_fusion/src/config/realsense_d435i/left.yaml
[ INFO] [1574410295.996333830]: reading paramerter of camera /home/admi/VINS_fusion/src/config/realsense_d435i/right.yaml
MULTIPLE_THREAD is 1
[ WARN] [1574410295.997425487]: waiting for image and imu...
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 
wait for imu ... 

rostopic list display:
admi@admi-virtual-machine:~$ rostopic list
/camera/accel/imu_info
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/gyro/imu_info
/camera/imu
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra1/image_rect_raw/compressed
/camera/infra1/image_rect_raw/compressed/parameter_descriptions
/camera/infra1/image_rect_raw/compressed/parameter_updates
/camera/infra1/image_rect_raw/compressedDepth
/camera/infra1/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra1/image_rect_raw/compressedDepth/parameter_updates
/camera/infra1/image_rect_raw/theora
/camera/infra1/image_rect_raw/theora/parameter_descriptions
/camera/infra1/image_rect_raw/theora/parameter_updates
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/infra2/image_rect_raw/compressed
/camera/infra2/image_rect_raw/compressed/parameter_descriptions
/camera/infra2/image_rect_raw/compressed/parameter_updates
/camera/infra2/image_rect_raw/compressedDepth
/camera/infra2/image_rect_raw/compressedDepth/parameter_descriptions
/camera/infra2/image_rect_raw/compressedDepth/parameter_updates
/camera/infra2/image_rect_raw/theora
/camera/infra2/image_rect_raw/theora/parameter_descriptions
/camera/infra2/image_rect_raw/theora/parameter_updates
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

closed 2019-11-25 by cugcsu

luxi-huang, 2020-02-11:
hello, I would like to ask how you solved "wait for imu" problem? I have the same problem.

Shame-fight, 2020-04-20:
I have the same problems,did you have solved it? @cugcsu

zhangjianglv, 2022-11-03:
@cugcsu I have the same problem（wait for imu）, did you solve it?

czp0204, 2023-04-23:
same problem here.

HonEYFIsh24, 2024-09-23:
Same problem. Anyone solved this plz?

camron-BHF, 2024-10-01:
Has anyone solved this problem lol. Running into it as well

Ethan-02, 2024-11-11:
have same problem with ros2

ssaigarimella, 2025-07-09:
@cugcsu has anyone solved this? Having the same issue