HKUST-Aerial-Robotics/VINS-Fusion #91: stack smashing detected

cugcsu, 2019-11-25:
Running vins-fusion with realsense d435i, an error occurred:
*** stack smashing detected ***: <unknown> terminated
[rvizvisualisation-1] process has died [pid 28755, exit code -6, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/admi/VINS_fusion/src/vins_estimator/../config/vins_rviz_config.rviz __name:=rvizvisualisation __log:=/home/admi/.ros/log/6a4a948a-0f59-11ea-ae6a-000c290466a6/rvizvisualisation-1.log].
log file: /home/admi/.ros/log/6a4a948a-0f59-11ea-ae6a-000c290466a6/rvizvisualisation-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
[image]

RozDavid, 2020-11-01:
Hey @cugcsu,

I am having the same error with the same device. Can you help me how you solved the problem?

Sadegh-Kalami, 2023-08-26:
Hello from another side of the earth :D @RozDavid @cugcsu 
I am having the same problem the first time I ran the code without any problem.
now I have this problem. Could u help me too ? pls

T-1009, 2024-02-01:
Have you solved it? I also encountered this problem

Sadegh-Kalami, 2024-02-01:
hey @T-1009  as far as I know due to low RAM memory Rviz is getting close.
if u can share ur system and OS details with us.