HKUST-Aerial-Robotics/VINS-Fusion #92: Camera pose frequency is half of the camera frame rate

jiqirenzhifu, 2019-11-29:
Thank you for your great work. I use MYNT camera to run vinsfusion and use rosbag to record corresponding topic packages. Why the topic frequency of the camera is half of the camera frame rate. How should I modify them to synchronize them? Thank you for any help
this is my rosbag info:
`     /mynteye/left/image_color      369 msgs    : sensor_msgs/Image     

             /vins_estimator/camera_pose    184 msgs    : nav_msgs/Odometry     

             /vins_estimator/margin_cloud   184 msgs    : sensor_msgs/PointCloud

             /vins_estimator/point_cloud    184 msgs    : sensor_msgs/PointCloud

`

Jwenda, 2019-12-03:
Hi, I have the same issue with D435i camera. The camera rate is 30Hz, but the odometry message is 15 Hz.

jiqirenzhifu, 2019-12-05:
@dvorak0

jiqirenzhifu, 2019-12-05:
Did you solve this problem @Jwenda

Jwenda, 2019-12-05:
Unfortunately not, I hope the authors could give us some reason. I might go through the code in following weeks.

shaozu, 2019-12-05:
Hi, [Here](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/0c3206941410723b8c62b7b0c6a6189b38ae7d99/vins_estimator/src/estimator/estimator.cpp#L180) is the code that halves the image frequency, and you may modify it in your project accordingly.

jiqirenzhifu, 2019-12-05:
Thank you very much, your answer was very helpful, good luck！

closed 2019-12-05 by jiqirenzhifu

djramakrishna, 2023-01-31:
Hi @shaozu  @jiqirenzhifu ! I'd like to know what was the reason to skip the frames in the first place ? while setting the multithreading flag to true ? Thank you.