HKUST-Aerial-Robotics/VINS-Fusion #94: [memory leakage] VINS-Fusion has memory leakage problem

zgxsin, 2019-12-04:
Hi, @qintonguav @shaojie @shaozu , I am running the latest VINS-Fusion [I don't change anything] over a dataset, and the memory just keeps increasing. You may see it [here](https://drive.google.com/open?id=1psmQ-13-LwV5rd-MymUQ0-ipgQwjWqwq). I also tested it by running the Euroc rosbags, it has memory leakage as well. I didn't fully check the code. I also observed that there a few memory-related issues in VINS-Mono. Maybe you forget to cherry-pick related commits to this repo?

sunbingfengPI, 2019-12-19:
Suggest to disable `DEBUG_IMAGE` in loop_fusion module, and it will thus release input image after keyframe construction.

otis-XJY, 2023-11-28:
> Maybe you should disable in loop_fusion module, and it will thus release input image after keyframe construction. @zgxsin`DEBUG_IMAGE`

I have the same question. Can you describe more specific please~

ZW-PRO, 2024-01-10:
I have the same question.  can you help me？ i dnot use loop_fusion ,only estimator.