HKUST-Aerial-Robotics/VINS-Fusion #95: With my own equipment, after configuring the camera parameters, the results are infinite

tianxinyu0509, 2019-12-05:
With my own equipment, after configuring the camera parameters, the results are infinite. If I double the fx,fy of the camera, I can get results, but it is not accurate.
infinite:
[image]
Causes an infinite parameter configuration:(This parameter is the result of calibration.)
[image]
If I double fx,fy, and give it a false parameter, I can output a pose, but it's not correct.
Can someone tell me why the correct parameters lead to infinite results?

tianxinyu0509, 2019-12-05:
I solved the problem of infinite results and re-calibrated the camera using the kannala-brandt model.Retest, found a bit of pose output, but not correct, should be the reason for body_T_cam0, how to re-calibrate this matrix?

ihamieh, 2020-09-11:
> I solved the problem of infinite results and re-calibrated the camera using the kannala-brandt model.Retest, found a bit of pose output, but not correct, should be the reason for body_T_cam0, how to re-calibrate this matrix?
@tianxinyu0509  can you share your yaml file? I'm having a similar issue and i'm not sure how to integrate fisheye to ORB-SLAM3? I made an old example work but it's causing me issues!