HKUST-Aerial-Robotics/VINS-Fusion #98: NaN values: Error in evaluating the ResidualBlock.

mzahana, 2019-12-20:
Hello.

I am trying to use VINS-FUSION with Realsense T265. I followed the following steps to run the setup.
* I recorded a rosbag of the two fisheye cameras at 5Hz rate.
* I did intrinsic camera calibration using Kalibr with `omni-radtan` models
* I recorded another bag with the imu topic only and used `imu_utils` to obtain IMU noise parameters
* I used Kalibr to calibrate cameras and IMU
* The resulting parameters were entered in the config file read by VINS, which looks like
```
%YAML:1.0

#common parameters
#support: 1 imu 1 cam; 1 imu 2 cam: 2 cam; 
imu: 1         
num_of_cam: 2  

imu_topic: "/camera/imu"
image0_topic: "/camera/fisheye1/image_raw"
image1_topic: "/camera/fisheye2/image_raw"
output_path: "/home/dji/output/"

cam0_calib: "left.yaml"
cam1_calib: "right.yaml"
image_width: 848
image_height: 800
   

# Extrinsic parameter between IMU and Camera.
estimate_extrinsic: 1   # 0  Have an accurate extrinsic parameters. We will trust the following imu^R_cam, imu^T_cam, don't change it.
                        # 1  Have an initial guess about extrinsic parameters. We will optimize around your initial guess.

# Left
body_T_cam0: !!opencv-matrix #cam0 coordinate under body coordinate
   rows: 4
   cols: 4
   dt: d
   data: [ -0.99991462,  0.00042657,  0.01306038,  0.01202619,
           -0.0004514,  -0.9999981,  -0.00189783,  0.00503566,
           0.01305954, -0.00190356,  0.99991291,  0.00686882,
           0., 0., 0., 1.]

body_T_cam1: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ -0.999985,    0.00234903,  0.00494827, -0.05189573,
           -0.00234375, -0.99999668,  0.00107182,  0.00487799, 
           0.00495077,  0.00106021,  0.99998718,  0.00708981,
           0., 0., 0., 1.]

#Multiple thread support
multiple_thread: 1

#feature traker paprameters
max_cnt: 150            # max feature number in feature tracking
min_dist: 30            # min distance between two features 
freq: 10                # frequence (Hz) of publish tracking result. At least 10Hz for good estimation. If set 0, the frequence will be same as raw image 
F_threshold: 1.0        # ransac threshold (pixel)
show_track: 0           # publish tracking image as topic
flow_back: 1            # perform forward and backward optical flow to improve feature tracking accuracy

#optimization parameters
max_solver_time: 0.04  # max solver itration time (ms), to guarantee real time
max_num_iterations: 8   # max solver itrations, to guarantee real time
keyframe_parallax: 10.0 # keyframe selection threshold (pixel)

#imu parameters       The more accurate parameters you provide, the better performance
acc_n: 0.018536          # accelerometer measurement noise standard deviation. #0.2   0.04
gyr_n: 0.00202783         # gyroscope measurement noise standard deviation.     #0.05  0.004
acc_w: 0.00061931 #0.001         # accelerometer bias random work noise standard deviation.  #0.002
[... 142 lines cut ...]

E1220 20:29:26.986863 25200 trust_region_minimizer.cc:72] Terminating: Residual and Jacobian evaluation failed.
[ INFO] [1576862966.994982779]: td 0.010765
time: 1576862966.950576, t: 0.018050 -0.024542 -0.025643 q: 0.683518 0.723155 -0.099245 -0.000940 
W1220 20:29:27.055917 25200 residual_block.cc:131] 

Error in evaluating the ResidualBlock.

There are two possible reasons. Either the CostFunction did not evaluate and fill all    
residual and jacobians that were requested or there was a non-finite value (nan/infinite)
generated during the or jacobian computation. 

Residual Block size: 6 parameter blocks x 2 residuals

For each parameter block, the value of the parameters are printed in the first column   
and the value of the jacobian under the corresponding residual. If a ParameterBlock was 
held constant then the corresponding jacobian is printed as 'Not Computed'. If an entry 
of the Jacobian/residual array was requested but was not written to by user code, it is 
indicated by 'Uninitialized'. This is an error. Residuals or Jacobian values evaluating 
to Inf or NaN is also an error. 
```

I need help to understand and solve this issue.

Cheers.

hridaybavle, 2020-04-20:
Hi, 

Where you able to solve the problem? I am facing the same issue.

mzahana, 2020-04-21:
@hridaybavle unfortunately, I sopped working on this for now. It's not yet resolved.

hridaybavle, 2020-04-21:
Hi @mzahana I was able to solve the issue. You can see this [pull request](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/pull/110).

zhangjianglv, 2022-11-14:
@mzahana Did you solve this problem later? I am also facing this issue.