HKUST-Aerial-Robotics/VINS-Fusion #106: Segmentation fault while running euroc_stereo_imu_config.yaml

lken01, 2020-03-13:
This might be a bit off, but I am getting segmentation fault while trying to run **euroc_stereo_imu_config.yaml** over the MH_01_easy dataset.

`rosrun vins vins_node src/VINS-Fusion/config/euroc/euroc_stereo_imu_config_copy.yaml 
[ INFO] [1584082013.857997258]: init begins

config_file: src/VINS-Fusion/config/euroc/euroc_stereo_imu_config_copy.yaml

USE_IMU: 1

IMU_TOPIC: /imu0

result path ~/output//vio.csv

[ WARN] [1584082013.869343952]:  fix extrinsic param 
camera number 1

[ INFO] [1584082013.869990193]: Synchronized sensors, fix time offset: 0
[ INFO] [1584082013.870170803]: ROW: 480 COL: 752 

 exitrinsic cam 0

 0.0148655  -0.999881  0.0041403
  0.999557  0.0149672  0.0257155
-0.0257744 0.00375619   0.999661
-0.0216401  -0.064677 0.00981073
set g       0       0 9.81007

[ INFO] [1584082013.871279533]: reading paramerter of camera src/VINS-Fusion/config/euroc/cam0_mei.yaml

MULTIPLE_THREAD is 1

[ WARN] [1584082013.871877933]: waiting for image and imu...
init first imu pose

averge acc 7.745210 -0.452196 -2.672312
init R0 

  0.330294    0.15748   0.930648
         0  -0.985983   0.166843
  0.943878 -0.0551073  -0.325664

[ WARN] [1584082017.894913265]: gyroscope bias initial calibration -0.00409239   0.0257547   0.0779806

Segmentation fault (core dumped)`

Help would be much appreciated. Thank you.

Mealoore, 2020-04-29:
the same problem

shaozu, 2020-04-29:
Hi, you can try [docker](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/master/README.md#7-docker-support) first in case of environment problem.

nickle-fang, 2021-06-19:
Hey, I have met the same problem. I solved it at last. It is because I use the Ubuntu 20.04 + ROS Noetic and my opecv conflicts with the ROS version.
You can try this repo which fix the problem in ROS noetic:[https://github.com/mrdavoodi/VINS-Fusion.git](url)

yubaoliu, 2022-04-16:
@nickle-fang I also met this error and I use the same environment as you. How you solved it? Thanks

Rick-Weng, 2022-10-06:
> Hey, I have met the same problem. I solved it at last. It is because I use the Ubuntu 20.04 + ROS Noetic and my opecv conflicts with the ROS version. You can try this repo which fix the problem in ROS noetic:[https://github.com/mrdavoodi/VINS-Fusion.git](url)哥们咋解决的阿，ROS20.04和Opencv版本冲突是要换成哪个版本opencv？

rocker12121, 2023-09-23:
> Hey, I have met the same problem. I solved it at last. It is because I use the Ubuntu 20.04 + ROS Noetic and my opecv conflicts with the ROS version. You can try this repo which fix the problem in ROS noetic:[https://github.com/mrdavoodi/VINS-Fusion.git](url)

Hello. the link you provided is not working. Can you please tell me how did you solve this error? Thank you.

vipslzyz, 2023-10-09:
same problem,is there any advise?

xuyuxiu83, 2024-12-05:
I encountered a similar problem in my workspace. My approach was to specify the path of cv_bridge while specifying the path of opencv. At the same time, I added the path of cv_bridge to include_directories and target_link_libraries. The files to be modified were loop_fusion/CMakeLists.txt and vins_estimator/CMakeLists.txt. Here is an example:

cmake_minimum_required(VERSION 2.8.3)

project(loop_fusion)

set(CMAKE_BUILD_TYPE "Release")

set(CMAKE_CXX_FLAGS "-std=c++11")

#-DEIGEN_USE_MKL_ALL")

set(CMAKE_CXX_FLAGS_RELEASE "-O3 -Wall -g")

find_package(catkin REQUIRED COMPONENTS
roscpp
std_msgs
nav_msgs
camera_models
# cv_bridge
roslib
) set(cv_bridge_DIR "/home/xyx/Downloads/3rdparties/vision_opencv-noetic/cv_bridge/install/install3413/share/cv_bridge/cmake") # Find cv_bridge package find_package(cv_bridge) set(OpenCV_DIR /home/xyx/Downloads/3rdparties/opencv-3.4.13/install/share/OpenCV) find_package(OpenCV 3.4 REQUIRED) find_package(Ceres REQUIRED) set(CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake) find_package(Eigen3) include_directories(${catkin_INCLUDE_DIRS} ${CERES_INCLUDE_DIRS} ${EIGEN3_INCLUDE_DIR} ${cv_bridge_INCLUDE_DIRS}) catkin_package() add_executable(loop_fusion_node src/pose_graph_node.cpp src/pose_graph.cpp src/keyframe.cpp src/utility/CameraPoseVisualization.cpp src/ThirdParty/DBoW/BowVector.cpp src/ThirdParty/DBoW/FBrief.cpp src/ThirdParty/DBoW/FeatureVector.cpp src/ThirdParty/DBoW/QueryResults.cpp src/ThirdParty/DBoW/ScoringObject.cpp src/ThirdParty/DUtils/Random.cpp src/ThirdParty/DUtils/Timestamp.cpp src/ThirdParty/DVision/BRIEF.cpp src/ThirdParty/VocabularyBinary.cpp ) target_link_libraries(loop_fusion_node ${catkin_LIBRARIES} ${OpenCV_LIBS} ${CERES_LIBRARIES} ${cv_bridge_LIBRARIES})

now everything seems to be OK~
[image]

xuyuxiu83, 2024-12-05:
I encountered a similar problem in my workspace. My approach was to specify the path of cv_bridge while specifying the path of opencv. At the same time, I added the path of cv_bridge to include_directories and target_link_libraries. The files to be modified were loop_fusion/CMakeLists.txt and vins_estimator/CMakeLists.txt. Here is an example:
```bash
cmake_minimum_required(VERSION 2.8.3)

project(loop_fusion)

set(CMAKE_BUILD_TYPE "Release")

set(CMAKE_CXX_FLAGS "-std=c++11")

#-DEIGEN_USE_MKL_ALL")

set(CMAKE_CXX_FLAGS_RELEASE "-O3 -Wall -g")

find_package(catkin REQUIRED COMPONENTS
roscpp
std_msgs
nav_msgs
camera_models

cv_bridge
roslib
) set(cv_bridge_DIR "/home/xyx/Downloads/3rdparties/vision_opencv-noetic/cv_bridge/install/install3413/share/cv_bridge/cmake") # Find cv_bridge package find_package(cv_bridge) set(OpenCV_DIR /home/xyx/Downloads/3rdparties/opencv-3.4.13/install/share/OpenCV) find_package(OpenCV 3.4 REQUIRED) find_package(Ceres REQUIRED) set(CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake) find_package(Eigen3) include_directories(${catkin_INCLUDE_DIRS} ${CERES_INCLUDE_DIRS} ${EIGEN3_INCLUDE_DIR} ${cv_bridge_INCLUDE_DIRS}) catkin_package() add_executable(loop_fusion_node src/pose_graph_node.cpp src/pose_graph.cpp src/keyframe.cpp src/utility/CameraPoseVisualization.cpp src/ThirdParty/DBoW/BowVector.cpp src/ThirdParty/DBoW/FBrief.cpp src/ThirdParty/DBoW/FeatureVector.cpp src/ThirdParty/DBoW/QueryResults.cpp src/ThirdParty/DBoW/ScoringObject.cpp src/ThirdParty/DUtils/Random.cpp src/ThirdParty/DUtils/Timestamp.cpp src/ThirdParty/DVision/BRIEF.cpp src/ThirdParty/VocabularyBinary.cpp ) target_link_libraries(loop_fusion_node ${catkin_LIBRARIES} ${OpenCV_LIBS} ${CERES_LIBRARIES} ${cv_bridge_LIBRARIES})

now everything seems to be OK~

xuyuxiu83, 2024-12-05:
I encountered a similar problem in my workspace. My approach was to specify the path of cv_bridge while specifying the path of opencv. At the same time, I added the path of cv_bridge to include_directories and target_link_libraries. The files to be modified were loop_fusion/CMakeLists.txt and vins_estimator/CMakeLists.txt. Here is an example:
```bash
cmake_minimum_required(VERSION 2.8.3)
project(loop_fusion)

set(CMAKE_BUILD_TYPE "Release")
set(CMAKE_CXX_FLAGS "-std=c++11")
#-DEIGEN_USE_MKL_ALL")
set(CMAKE_CXX_FLAGS_RELEASE "-O3 -Wall -g")

find_package(catkin REQUIRED COMPONENTS
    roscpp
    std_msgs
    nav_msgs
    camera_models
#    cv_bridge
    roslib
    )
set(cv_bridge_DIR "/home/xyx/Downloads/3rdparties/vision_opencv-noetic/cv_bridge/install/install3413/share/cv_bridge/cmake")
# 查找 cv_bridge 包
find_package(cv_bridge)

set(OpenCV_DIR /home/xyx/Downloads/3rdparties/opencv-3.4.13/install/share/OpenCV)
find_package(OpenCV 3.4 REQUIRED)

find_package(Ceres REQUIRED)

set(CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake)
find_package(Eigen3)

include_directories(${catkin_INCLUDE_DIRS} ${CERES_INCLUDE_DIRS}  ${EIGEN3_INCLUDE_DIR} ${cv_bridge_INCLUDE_DIRS})

catkin_package()

add_executable(loop_fusion_node
    src/pose_graph_node.cpp
    src/pose_graph.cpp
    src/keyframe.cpp
    src/utility/CameraPoseVisualization.cpp
    src/ThirdParty/DBoW/BowVector.cpp
    src/ThirdParty/DBoW/FBrief.cpp
    src/ThirdParty/DBoW/FeatureVector.cpp
    src/ThirdParty/DBoW/QueryResults.cpp
    src/ThirdParty/DBoW/ScoringObject.cpp
    src/ThirdParty/DUtils/Random.cpp
    src/ThirdParty/DUtils/Timestamp.cpp
    src/ThirdParty/DVision/BRIEF.cpp
    src/ThirdParty/VocabularyBinary.cpp
    )

target_link_libraries(loop_fusion_node ${catkin_LIBRARIES}  ${OpenCV_LIBS} ${CERES_LIBRARIES} ${cv_bridge_LIBRARIES})
```

I think this issue hax been perfectly solved. Now everything seems to be OK~
[image]

xuyuxiu83, 2024-12-05:
I encountered a similar problem in my workspace. My approach was to specify the path of cv_bridge while specifying the path of opencv. At the same time, I added the path of cv_bridge to include_directories and target_link_libraries. The files to be modified were loop_fusion/CMakeLists.txt and vins_estimator/CMakeLists.txt. Here is an example:
```bash
cmake_minimum_required(VERSION 2.8.3)
project(loop_fusion)

set(CMAKE_BUILD_TYPE "Release")
set(CMAKE_CXX_FLAGS "-std=c++11")
#-DEIGEN_USE_MKL_ALL")
set(CMAKE_CXX_FLAGS_RELEASE "-O3 -Wall -g")

find_package(catkin REQUIRED COMPONENTS
    roscpp
    std_msgs
    nav_msgs
    camera_models
#    cv_bridge
    roslib
    )
set(cv_bridge_DIR "/home/xyx/Downloads/3rdparties/vision_opencv-noetic/cv_bridge/install/install3413/share/cv_bridge/cmake")
# 查找 cv_bridge 包
find_package(cv_bridge)

set(OpenCV_DIR /home/xyx/Downloads/3rdparties/opencv-3.4.13/install/share/OpenCV)
find_package(OpenCV 3.4 REQUIRED)

find_package(Ceres REQUIRED)

set(CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake)
find_package(Eigen3)

include_directories(${catkin_INCLUDE_DIRS} ${CERES_INCLUDE_DIRS}  ${EIGEN3_INCLUDE_DIR} ${cv_bridge_INCLUDE_DIRS})

catkin_package()

add_executable(loop_fusion_node
    src/pose_graph_node.cpp
    src/pose_graph.cpp
    src/keyframe.cpp
    src/utility/CameraPoseVisualization.cpp
    src/ThirdParty/DBoW/BowVector.cpp
    src/ThirdParty/DBoW/FBrief.cpp
    src/ThirdParty/DBoW/FeatureVector.cpp
    src/ThirdParty/DBoW/QueryResults.cpp
    src/ThirdParty/DBoW/ScoringObject.cpp
    src/ThirdParty/DUtils/Random.cpp
    src/ThirdParty/DUtils/Timestamp.cpp
    src/ThirdParty/DVision/BRIEF.cpp
    src/ThirdParty/VocabularyBinary.cpp
    )

target_link_libraries(loop_fusion_node ${catkin_LIBRARIES}  ${OpenCV_LIBS} ${CERES_LIBRARIES} ${cv_bridge_LIBRARIES})
```
At the same time, it is necessary to strictly note that cv_bridge and opencv here should be compiled accordingly, and the original version of the system cannot be used.
I think this issue hax been perfectly solved. Now everything seems to be OK~
[image]