HKUST-Aerial-Robotics/VINS-Fusion #109: Using the constant 460

AhmedElsafy, 2020-04-02:
https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/0c3206941410723b8c62b7b0c6a6189b38ae7d99/vins_estimator/src/initial/solve_5pts.cpp#L204

I have been debugging around an issue I have with not enough parallax, and I have noticed this constant 460 been used multiple times when calculating the fundamental matrix for the epipolar threshold and also when doing a check for parallax at the estimator as well in https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/blob/0c3206941410723b8c62b7b0c6a6189b38ae7d99/vins_estimator/src/estimator/estimator.cpp#L807 

Is there a reason for that ?

tangxiangshi, 2021-05-13:
460 is the focal length

zhh2005757, 2022-04-05:
I think it should have been the value identified to your configuration file with focal length