HKUST-Aerial-Robotics/VINS-Fusion #114: Camera frame moves crazy in RVIZ with D435i

mhaboali, 2020-05-13:
Thanks for sharing this great project!

When I tried it with RealSense D435i using your docker method and your d435i.yaml config file, it worked but while moving it very little movement I saw it moved crazy in RVIZ and didn't stop at all as shown in the attached

[image]

Note: I didn't change anything except for topics names and paths

Shame-fight, 2020-05-15:
As far as I know, many people have the same problem, including me. This phenomenon is very strange. Different cameras and different devices have encountered the same problem. I suspect that it is an algorithm problem or initialization. Troubled for a while, and do not know the solution. If you know some reasons or solve this problem, please let me know.

xdtl, 2020-05-15:
Same here. It works great on MH_01_easy.bag. But when I use my own data, I got crazy camera trajectories. And every time the result is different with exactly the same setting. Could someone please give suggestions about how to identify and correct the problem? Thanks!

jagauthier, 2020-05-28:
I'm also having this issue from a rosbag file I created with an realsense R200.

shirbarzel, 2020-07-20:
Any updates? I'm having the same problem?

hanzheteng, 2020-07-30:
Perhaps you need to calibrate your IMU first

shirbarzel, 2020-07-30:
> Perhaps you need to calibrate your IMU first

I did it using the rs-imu-calibration tool by realsense, @hanzheteng have you managed to get a stable VIO with the D435i and VINS-FUSION?

dssdyx, 2021-04-13:
I use realsense d455 to run VINS Fusion and get great result. I think d435i and d455 have similar config. There are some important  points deciding whether the system is stable or not.

1.  Extrinsic between imu and cameras. I think stable vins system must depend on good extrinsic and we can use kalibr to solve it. The result of kalibr is Tci , but in vins config, body_T_cam seems to mean Tic.
2. Imu parameters. I used Allan variance to estimate imu noise and random walk. Actually, it's the main reason why I always get crazy result. Using default imu parameters such as 0.1, 0.001 will get good result. I think it's because when I calibrated imu, the system was static without vibration,which was different in moving situation. Imu noise and random walk should be increased in fact.
3. Synchronization problem. This is a little chaotic problem, because different realsense versions and firmwares have different situation.I now use librealsense 2.42, realsense-ros 2.2.22, which is ok. Also, remember calibrate time offset between cameras and imu in Kalibr using "--time-calibration" flag

Camilochiang, 2022-10-25:
I used  a RS415. I believe that the most important point, as mentioned by @dssdyx is the sync. After calibration with Klibr I got a value of 0.3
Here with td:0 (it fail)
[image]

Here with td:0.3
[image]