HKUST-Aerial-Robotics/VINS-Fusion #116: vins-fusion, I have errors.

leehyun0404, 2020-05-29:
Hello I am korean.
I am not good at English.
I hope you understand.

I faced a problem while running vins fusion.
Will you help me?

 OS type : 64bit ubuntu 16.04 ROS kinetic
processor : Intel core i7-4770HQ @ 2.20GHz x8

realsense-ros 2.2.13
librealsense 2.33 .1
Eigen 3.3.7
Ceres solver 1.14.0
Realsense D435I

I opened four terminals
window1
: roslaunch vins vins_rviz.launch

window2
: roslaunch realsense2_camera rs_camera.launch

window3
: rosrun vins vins node ~/catkin_ws/src/VINS-Fusion/config/realsense_d435i/realsense_stereo_imu_config.yaml

I have a problem Will you help me? Please...

suyu13020027161, 2020-07-13:
i have the same problem, i use the gazebo simulator, but vins doesn't seem to work properly.

Pallav1299, 2021-10-05:
I tested VINS-FUSION with gazebo and real-world. Following are the results:
**Real world:**
Decent performance with Realsense D435i.

**Gazebo:**
Stereo mode is giving decent results while, in VI Stereo the performance degrades.

Poor VI-SLAM results with VINS Fusion in gazebo might be due to the following:
1. Bad data from IMU plugin. I configured the plugin to output data with zero noise and sensor bias. But still the noise exists. I verified this by testing with imu_utils for finding the sensor noise and bias. I tested it with default gazebo plugins and hector_gazebo_imu_plugin.
2. I am also consistently getting the following optimizer warnings/errors:
```
Error in evaluating the ResidualBlock.

There are two possible reasons. Either the CostFunction did not evaluate and fill all    
residual and jacobians that were requested or there was a non-finite value (nan/infinite)
generated during the or jacobian computation. 

Residual Block size: 5 parameter blocks x 2 residuals

For each parameter block, the value of the parameters are printed in the first column   
and the value of the jacobian under the corresponding residual. If a ParameterBlock was 
held constant then the corresponding jacobian is printed as 'Not Computed'. If an entry 
of the Jacobian/residual array was requested but was not written to by user code, it is 
indicated by 'Uninitialized'. This is an error. Residuals or Jacobian values evaluating 
to Inf or NaN is also an error.  

Residuals:             -nan         -nan 

Parameter Block 0, size: 7

     1.07506 |     -43.8169      9.63801 
    -1.48701 |     -62.1792       1.2691 
    0.226267 |    -0.881051     -62.0186 
  0.00928231 |      50.6846      224.495 
  0.00369336 |     -35.7264      325.458 
  0.00283941 |     -477.187      36.9214 
    0.999946 |            0            0 

Parameter Block 1, size: 7

     1.12702 |      43.8169     -9.63801 
    -1.58163 |      62.1792      -1.2691 
    0.236191 |     0.881051      62.0186 
  0.00981494 |     -50.0767     -218.493 
   0.0038356 |      35.6784     -322.279 
  0.00264367 |      469.772     -35.6784 
    0.999941 |            0            0 

Parameter Block 2, size: 7

        0.12 |   -0.0241753   -0.0172021 
           0 |    0.0172549    0.0621977 
           0 |   -0.0538629  -0.00277532 
         0.5 |    0.0544573     -3.17951 
        -0.5 |       7.4171      -1.2355 
         0.5 |     0.607914      6.00278 
        -0.5 |            0            0 

Parameter Block 3, size: 1

         0.2 |     -18.0016      1.16311 

Parameter Block 4, size: 1

           0 | Not Computed  Not Computed  

E1005 11:43:57.926491 27988 trust_region_minimizer.cc:72] Terminating: Residual and Jacobian evaluation failed.
```

Has someone been able to get it working in gazebo?

MuhammadAhmedMasood, 2022-04-24:
> 

Hi ! Can you tell how did you do it using Gazebo. I am also have huge drift on rviz.

zhangjianglv, 2022-11-16:
@Pallav1299 hi,I ran into the same problem as you, did you solve it?

leehyun0404, 2022-11-17:
Hi, 
It was a long time ago.
i didn't solve problem.

I will give you a reference site.
youtube : Mason Eungchang

send he a mail.
good luck  ~