HKUST-Aerial-Robotics/VINS-Fusion #124: Running on the TUN VI dataset

KopanevPavel, 2020-07-23:
Hello) Had someone successfully run VINS-Fusion on the [TUM VI](https://vision.in.tum.de/data/datasets/visual-inertial-dataset) dataset? I noticed that there were [config files](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/tree/master/config/tum) for that in the [VINS-Mono repo](https://github.com/HKUST-Aerial-Robotics/VINS-Mono). I tested VINS-Mono with the TUM VI dataset and everything was ok. But just transferring the data in VINS-Fusion didn't help and trajectory was wrong.

Also are the following flags in the [config file](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/config/tum/tum_config.yaml) of VINS-Mono still relevant here?:
*equalize: 1            
fisheye: 1*

hobbeshunter, 2020-10-16:
I gave it a try at https://github.com/hobbeshunter/VINS-Fusion/commit/82e2582630186da83a70803c1efe48db74c6cc27 but it did not work. I would very much like to know what I did wrong.

otakudj, 2026-10-16:
I have adapt my own ego data, which has the same camera type tum-vi has,  to VINS-Fusion these days and solve the tum-vi adaption problem, you guys can give it a try at https://github.com/otakudj/VINS-Fusion/commit/833e76663082782dfa4b3fa9fdd77de4418fa5dc.