HKUST-Aerial-Robotics/VINS-Fusion #131: vins fusion optimization nan

BeatScherrer, 2020-09-07:
I am trying to run vins fusion with my data from gazebo but sometimes receive the following errors (see log file)
[vins_fusion_log.txt](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/files/5182260/vins_fusion_log.txt)

I am using the following configuration:
[heap_fusion_config_sim.txt](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/files/5182268/heap_fusion_config_sim.txt)

The error does not always occur but if it does not occur loop closure does not work. Funnily enough if I run a dataset with real data the optimization and also loop closure is working properly.

Pallav1299, 2021-09-15:
@BeatScherrer, I am facing the same issue. Any updates?

BeatScherrer, 2021-09-15:
> @BeatScherrer, I am facing the same issue. Any updates?

IIRC the issue was feature correspondences as illustrated above. Try to visualize the matched features. This error is most likely due to something going terribly wrong due to these correspondences manifesting in NaNs

Pallav1299, 2021-09-15:
I am not getting stable results in gazebo simulation, with the optimization error log popping up. On the other hand, it's working pretty well in real world. Any ideas, why is this happening? Have you explored this?

BeatScherrer, 2021-09-15:
> > IIRC the issue was feature correspondences as illustrated above. Try to visualize the matched features. This error is most likely due to something going terribly wrong due to these correspondences manifesting in NaNs
> 
> I am not getting stable results in gazebo simulation, with the optimization error log popping up. On the other hand, it's working pretty well in real world. Any ideas, why is this happening? Have you explored this?

Did you make sure your feature correspondences are reasonable? I explained why I received this error in the edit and therefore I would check if your environment is appropriate for the algorithms used. Feature matching can be difficult in simulations whereas it's usually working very well in the real world.

Pallav1299, 2021-09-15:
> Did you make sure your feature correspondences are reasonable? I explained why I received this error in the edit and therefore I would check if your environment is appropriate for the algorithms [...]

@BeatScherrer, Here is the [world](https://github.com/HKPolyU-UAV/E2ES/blob/master/gazebo/worlds/indoor20x20.world) that I used in gazebo, in case you want to have a look. Please let me know your views about the same.

BeatScherrer, 2021-09-15:
#
> @BeatScherrer, Here is the [world](https://github.com/HKPolyU-UAV/E2ES/blob/master/gazebo/worlds/indoor20x20.world) that I used in gazebo, in case you want to have a look. Please let me know your views about the same.

I don't have anything set up to be able to test this and honestly don't want to take the time. If it's not the environment, which you can easily verify by taking one of these worlds https://github.com/ethz-asl/rotors_simulator/tree/master/rotors_gazebo/worlds, I don't know what your problem might be. The optimization is not too explicit about its error in VINS-Fusion AFAIK.

Pallav1299, 2021-09-16:
@BeatScherrer Thanks for the help.