HKUST-Aerial-Robotics/VINS-Fusion #134: Segmentation Fault running

tainibeibi1337, 2020-09-16:
Hello!

After compiling VINS-Fusion successfully, on a x86_64 machine, I am trying to execute it using Euroc config file, as so:
`rosrun vins vins_node ~/catkin_ws/src/VINS_Fusion/config/euroc/euroc_mono_imu_config.yaml`

Execution results in **Segmentation fault**:
`$ rosrun vins vins_node ~/catkin_ws/src/VINS_Fusion/config/euroc/euroc_mono_imu_config.yaml 
[ INFO] [1600243514.529123941]: init begins
config_file: /home/user/catkin_ws/src/VINS_Fusion/config/euroc/euroc_mono_imu_config.yaml
Segmentation fault`

I have `opencv-4.4` built from source.

When trying to debug the executable using gdb, I stumbled upon this error:
`Thread 1 "vins_node" received signal SIGSEGV, Segmentation fault.
__strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:93
93      ../sysdeps/x86_64/multiarch/strlen-avx2.S: No such file or directory.`
I wasn't able to identify the root cause of the problem myself.

What could be the problem? Could it be the version of OpenCV?
I have a Jetson Xavier with `opencv-4.3` installed on it, and Vins works fine there.

Would be glad for some help,
Thank you for your work!

001SCH, 2020-11-02:
I also encountered the same problem, using gdb debugging to find the problem in the rosNodeTest.cpp file readParameter this function, do you have a solution?

robin-shaun, 2021-03-07:
@tainibeibi1337 I changed opencv version from opencv4.1.1 to opencv3.2.0 and solved the error. For @001SCH , I think it is the wrong path of config file that cause the problem.

zhongguanLiu, 2023-04-11:
hi, have you solve this problem?
my opencv version(pkg-config --modversion opencv) is 3.4.16, but i also meet the same fault, i dont konw how to solve it .
and my ubutun version is 20.04