HKUST-Aerial-Robotics/VINS-Fusion #137: Throw img1 with stereo imu config.yaml

CeccAnd, 2020-10-24:
Hello

I had followed all procedure here https://www.programmersought.com/article/62434283529/ to install and use VINS FUSION stereo+IMU on camera T265.

When I am running rosrun vins vins_node ~/catkin_ws/src/VINS-Fusion/config/realsense_t265review/stereo_imu.yaml, the result is somethig like below, plenty of throw img0 and throw img1:

[ INFO] [1603534498.335273765]: td 0.000000
time: 1603534497.714195, t: 0.000245 -0.000871 0.000009 q: 0.981834 -0.189737 -0.001241 -0.000461 
[ INFO] [1603534498.445917502]: td 0.000000
time: 1603534497.780895, t: 0.000275 -0.001510 -0.000055 q: 0.981818 -0.189820 -0.001154 -0.000235 
throw img0
throw img1

Therefore if I am going to launch roslaunch vins, the camera is stuck.

Any idea to solve? 

Thanks

rptamin, 2020-12-21:
I am having similar issues. I am running VINS in a docker with camera feed from AirSim. I was running stereo only config file.

terrense, 2021-01-13:
please tell me how to solve this question~~

MinnDevelopment, 2021-12-05:
I'm running into the same problem, any way to get more info on this error?

CeccAnd, 2021-12-06:
Unfortunately not. I stopped the project.

seqwalt, 2022-05-30:
I think this error occurs because the stereo images do not satisfy the 0.003 second sync tolerance. See lines 85 - 95 in vins_estimator/src/rosNodeTest.cpp. These lines of code are:

// 0.003s sync tolerance
if(time0 < time1 - 0.003)
{
    img0_buf.pop();
    printf("throw img0\n");
}
else if(time0 > time1 + 0.003)
{
    img1_buf.pop();
    printf("throw img1\n");
}

So basically, just make sure the two images (from simulator or camera) are published at the same time.