HKUST-Aerial-Robotics/VINS-Fusion #139: How to fix features as constant and do motion-only BA

You-zeyi, 2021-01-06:
The motion only BA is mentioned in vins-mono paper but I cant see it in the code(vins-mono and vins-fusion). The para_feature was not added to parameterBlock like all other parameters, at least not explicitly, which is puzzling to me. Can I apply SetParameterBlockConstant to para_feature? Should I apply AddParameterBlock first to para_feature before SetParameterBlockConstant.
I would appreciate it if An example code can be provided.

closed 2021-01-07 by You-zeyi

aphasiayc, 2022-09-28:
> The motion only BA is mentioned in vins-mono

I'm also confused about this para_feature issue... So any update on this thread?  Thanks!

YoungsooH, 2024-11-11:
Did you find the motion only BA part? I can't find that part too...