HKUST-Aerial-Robotics/VINS-Fusion #147: Optimization nan at vins_estimator

takmin, 2021-03-15:
I encountered similar error to #131 without loop closure.

I tried to run vins_estimator with l515 with the following realsense launch file and vins-fusion config file.
These parameters were calibrated with kalibr.

realsense launch file:
```
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="320"/>
  <arg name="depth_height"        default="240"/>
  <arg name="enable_depth"        default="false"/>

  <arg name="infra_width"        default="640"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>

  <arg name="color_width"         default="960"/>
  <arg name="color_height"        default="540"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"          default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="400"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"           default="true"/>
  <arg name="align_depth"           default="true"/>

  <arg name="filters"               default=""/>
  <arg name="clip_distance"         default="-2"/>
  <arg name="linear_accel_cov"      default="0.01"/>
  <arg name="initial_reset"         default="false"/>
  <arg name="unite_imu_method"      default="linear_interpolation"/>
  <arg name="hold_back_imu_for_frames"      default="true"/>
  
  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
      <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye"           value="$(arg enable_fisheye)"/>

      <arg name="depth_width"              value="$(arg depth_width)"/>
[... 199 lines cut ...]

Parameter Block 2, size: 3

    -20.1169 |      4.66792     -5.18916 
     10.5083 |      2.79882     -6.69498 
 3.46882e-12 |       543.87      -286.32 

E0314 12:37:06.469475 14526 trust_region_minimizer.cc:73] Terminating: Residual and Jacobian evaluation failed.
[ WARN] [1615693026.469850170]: gyroscope bias initial calibration -0.000474674  -0.00118822 -0.000186036
[ INFO] [1615693026.470229516]: misalign visual structure with IMU
[ WARN] [1615693026.598303594]: gyroscope bias initial calibration -nan -nan -nan
[ INFO] [1615693026.714800860]: Initialization finish!
[ INFO] [1615693026.714975415]: td 0.000000
time: 1615300596.271713, t: nan nan -nan q: nan -nan -nan -nan 
[ WARN] [1615693026.726596187]: Marker 'CameraPoseVisualization/1': invalid point[0] (nan, nan, nan)
[ WARN] [1615693026.726629238]: Marker 'CameraPoseVisualization/1': invalid point[1] (nan, nan, nan)
[ WARN] [1615693026.726644708]: Marker 'CameraPoseVisualization/1': invalid point[2] (nan, nan, nan)
[ WARN] [1615693026.726659906]: Marker 'CameraPoseVisualization/1': invalid point[3] (nan, nan, nan)
.
.
.
```

It seems something wrong around imu, but I have no idea.
I appreciate if anybody gives me any hints to solve this problem.

takmin, 2021-04-20:
I carefully traced the error and found the reason:

In FeatureTracker::undistortedPts(), tracking points are projected to new coordinates which eliminates lens distortion effect., however some points are projected to inf.
I set "distortion_parameters:" to zero in camera.txt, then it worked.

takmin, 2021-04-26:
This error is caused in PinholeCamera::liftProjective().
At line 501 in PinholeCamera.cc, parameter d_u is amplified by calling distortion() function recursively.
I could avoid error by changing the number of iteration to 1.
I don't know why the distortion() function is called iteratively, so I wonder if it is the right solution.

ahjcahkl, 2024-01-11:
I also encountered a similar problem. I set the iteration count of this function to 1 and it can run normally, up to a maximum of 3. However, if I add it up again, there will be an issue with nan, which I am very confused about. I have recalibrated the parameters of the IMU and camera, and successfully performed Kalibr joint calibration. Do you have any idea about this? I would greatly appreciate it if you could answer me.

asdkakad, 2025-03-17:
have you solve the problem,i meet the same in my d435i