HKUST-Aerial-Robotics/VINS-Fusion #150: 请问您使用DJI的A3 有和相机进行时钟硬同步吗？请问如何实现A3与其他传感器的时钟硬同步

HuangDihe, 2021-03-31:
请问您使用DJI的A3 有和相机进行时钟硬同步吗？请问如何实现A3与其他传感器的时钟硬同步

killnice, 2022-04-18:
你看他的论文里的实物图，很明显是用了硬件同步，用脉冲同步了CAM和imu

killnice, 2022-06-07:
> 请问您解决这个同步问题了吗？

解决了，我用单片机写了脉冲输出程序，用方波上升沿触发相机和IMU

hanlinniu, 2022-06-07:
> > 请问您解决这个同步问题了吗？
> 
> 解决了，我用单片机写了脉冲输出程序，用方波上升沿触发相机和IMU

您好，我用的是intel realsense D435，和飞控的IMU，这样也能用单片机触发相机和IMU吗？

killnice, 2022-06-08:
> > > 请问您解决这个同步问题了吗？
> > 
> > 
> > 解决了，我用单片机写了脉冲输出程序，用方波上升沿触发相机和IMU
> 
> 您好，我用的是intel realsense D435，和飞控的IMU，这样也能用单片机触发相机和IMU吗？

D435不能被外部触发，你想办法做软件上的时间戳对齐吧

JasonWang9805, 2023-06-15:
> 

我看到官方提供realsense硬件同步例子，你之前了解过这个吗
https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#introduction

JasonWang9805, 2023-06-15:
> > > > 请问您解决这个同步问题了吗？
> > > 
> > > 
> > > 解决了，我用单片机写了脉冲输出程序，用方波上升沿触发相机和IMU
> > 
> > 
> > 您好，我用的是Intel Realsense D435，和飞控的IMU，这样也能用单片机触发相机和IMU吗？
> 
> D435不能被外部触发，你想办法做软件上的时间戳对齐吧
我看到官方提供realsense硬件同步例子，你之前了解过这个吗https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#introduction

JasonWang9805, 2023-06-15:
> > > >> 请问您解决这个同步问题了吗？
> > > >
> > > >
> > >> 解决了，我用单片机写了脉冲输出程序，用方波上升沿触发相机和IMU
> > >
> > >
> >> 您好，我用的是Intel Realsense D435，和飞控的IMU，这样也能用单片机触发相机和IMU吗？
> >
>> D435不能被外部触发，你想办法做软件上的时间戳对齐吧
>
> 我看到官方提供realsense硬件同步例子，你之前了解过这个吗
> https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#introduction