HKUST-Aerial-Robotics/VINS-Fusion #151: VINS-FUSION on ZED Camera

truncs, 2021-04-09:
Hi,

I was trying to run VINS-Fusion on zed cameras but the output seems unstable (trajectory is random). I am attaching the screenshot of my results and the config files that I use 

[image]

Config File
```
%YAML:1.0

#common parameters
#support: 1 imu 1 cam; 1 imu 2 cam: 2 cam; 
imu: 1
num_of_cam: 2

imu_topic: "/zed2/zed_node/imu/data_raw"
image0_topic: "/zed2/zed_node/left_raw/image_raw_color"
image1_topic: "/zed2/zed_node/right_raw/image_raw_color"
output_path: "../output/"

cam0_calib: "left.yaml"
cam1_calib: "right.yaml"
image_width: 1280
image_height: 720
   

# Extrinsic parameter between IMU and Camera.
estimate_extrinsic: 1   # 0  Have an accurate extrinsic parameters. We will trust the following imu^R_cam, imu^T_cam, don't change it.
                        # 1  Have an initial guess about extrinsic parameters. We will optimize around your initial guess.

body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [0.0023661383021121862, -0.9999852376642965, 0.004891405014821881, 0.018520139160040982,
          -0.001778892596313919, -0.004895620035355042, -0.9999864341307854, 0.005185516233583466,
          0.9999956184557357, 0.0023574049192230384, -0.0017904500497825104, 0.03995238963352001,
          0.0, 0.0, 0.0, 1.0]

body_T_cam1: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [0.00916668776984686, -0.9999508579504861, 0.003775383347028649, -0.10031401932563686,
           0.000799258388630264, -0.0037682139625440647, -0.9999925808472585, 0.005421274724173588,
           0.9999576656145792, 0.009169637267600617, 0.0007646770705903583, 0.040830056141113125,
           0.0, 0.0, 0.0, 1.0]

#Multiple thread support
multiple_thread: 1
use_gpu: 1
use_gpu_acc_flow: 1

#feature traker paprameters
max_cnt: 150            # max feature number in feature tracking
min_dist: 30            # min distance between two features 
freq: 0                # frequence (Hz) of publish tracking result. At least 10Hz for good estimation. If set 0, the frequence will be same as raw image 
F_threshold: 1.0        # ransac threshold (pixel)
show_track: 1           # publish tracking image as topic
flow_back: 1            # perform forward and backward optical flow to improve feature tracking accuracy

#optimization parameters
max_solver_time: 0.04  # max solver itration time (ms), to guarantee real time
max_num_iterations: 8   # max solver itrations, to guarantee real time
keyframe_parallax: 10.0 # keyframe selection threshold (pixel)

#imu parameters       The more accurate parameters you provide, the better performance
acc_n: 2.3132361656574518e-02        # accelerometer measurement noise standard deviation. 
gyr_n: 2.1582698665500383e-03         # gyroscope measurement noise standard deviation.     
acc_w: 3.9710211871507773e-04       # accelerometer bias random work noise standard deviation.  
gyr_w: 7.8786569685973787e-06     # gyroscope bias random work noise standard deviation.     
g_norm: 9.81     # gravity magnitude

#unsynchronization parameters
estimate_td: 1   # online estimate time offset between camera and imu
td: 0.017919297596632126                             # initial value of time offset. unit: s. readed image clock + td = real image clock (IMU clock)

#loop closure parameters
load_previous_pose_graph: 0        # load and reuse previous pose graph; load from 'pose_graph_save_path'
pose_graph_save_path: "../output/pose_graph/" # save and load path
save_image: 1                   # save image in pose graph for visualization prupose; you can close this function by setting 0 

```
Using similar parameters for s-msckf works. Let me know pointers on how to debug.

lbw0502, 2021-05-18:
Hi, 
I have also encountered a similar problem(trajectory drifts a lot) when using zed2 camera. Have you figured out the reason?

MeisonP, 2021-07-21:
@lbw0502 Have you figured out the reason?

truncs, 2021-07-21:
I haven't figured out the reason for ZED specifically since I moved over to a different camera sensor and for that particular sensor I could solve the same problem by using the MEI camera model.

closed 2021-08-06 by truncs

mikodham, 2022-03-15:
Yes. I have used the calibrator here as well to find the camera parameters for the 2 cameras, 
thttps://support.stereolabs.com/hc/en-us/articles/360012749113-How-can-I-use-Kalibr-with-the-ZED-Mini-camera-in-ROS-
But the trajectory is also random. Does anyone know why?