HKUST-Aerial-Robotics/VINS-Fusion #157: Extreme drift and extrinsic parameters

Amer-Alhamvi, 2021-07-07:
**Hope you all fine,**

I am experiencing extreme drifts (sometimes even when not moving the camera).
My setup is ZED2 camera using the built in imu.
The drifts occur in all three  modes (mono, stereo, stereo+imu)
For camera calibration, I use the factory calibration from the manufacturer (cameras are individually calibrated)
But my problem is that I can't seem to find the correct extrinsic parameters for the imu cameras transformations (so I believe) 

I tried the factory extrinsic parameters for cam0 that prints when running the zed ros wrapper,
I tried calibrating with kalibr and using T_cam_imu AND T_imu_cam for each camera,
And I even tried to just use the baseline for the stereo set (in stereo mode) and ignoring the imu,

Yet still I can't get rid of the extreme drifts occurring.
 **Thank you,**

wangtsiao, 2021-07-23:
hi @Amer-Alhamvi , i face the same problem as you described.

i think the drift is because there is no good extrinsic parameter between imu and camera. what's more, you can't move you device too fast, It will cause the program to capture keyframe failed.

Amer-Alhamvi, 2021-08-17:
I found the problem, although the cameras' intrinsic parameters from the factory calibration are good, the imu-cameras parameters are completely "different".
I don't have much experience so I can't say if they are wrong or I don't understand what they stand for.

Anyway, I calibrated the camera again using Kalibr but with more care and used the inverses of T_cam_imu matrices for each camera which seems to work fine.
Also, sudden movements do not cause the system to diverge.

closed 2021-08-17 by Amer-Alhamvi

bikk7, 2021-09-13:
> I found the problem, although the cameras' intrinsic parameters from the factory calibration are good, the imu-cameras parameters are completely "different".
> I don't have much experience so I can't [...]
Which two topics do you use to calibrate cameras? How much reprojection error do you get using Kalibr to calibrate the cameras and imu-cameras parameters? I used to get more than 0.35 while Kalibr thinks the reprojection error between 0.1 and 0.2 is good.

bikk7, 2021-09-13:
Hi @Amer-Alhamvi,  Which two topics do you use to calibrate cameras?  How much reprojection error do you get using Kalibr to calibrate the cameras and imu-cameras parameters?  I used to get more than 0.35 while Kalibr thinks the reprojection error between 0.1 and 0.2 is good.