HKUST-Aerial-Robotics/VINS-Fusion #158: symbol lookup error

pzx-Id, 2021-07-25:
/home/lab/catkin_ws/devel/lib/vins/vins_node: symbol lookup error: /home/lab/catkin_ws/devel/lib/libcamera_models.so: undefined symbol: _ZN6google21kLogSiteUninitializedE

yangy4444, 2022-05-25:
I have the same problem。
Is your problem solved?

colinleng, 2022-09-02:
How to fix this, I have the same problem

shaoxh, 2022-09-11:
This error occurred when I use vins on KITTI dataset. Have no idea of the reason. Let me know if anyone tickle it.

huynhducmink, 2022-10-28:
@shaoxh @colinleng @yangy4444 @pzx-Id 
Try to rebuild Ceres 2.1.0 stable release then rebuild the workspace. That should fix the issues.
I believe the error is comming from an update of the libgoogle-glog-dev package, which Ceres depends on.