HKUST-Aerial-Robotics/VINS-Fusion #167: VINS-fusion on Ubuntu 20.04

peppegti, 2021-11-16:
Hello all,
I try to use VINS-fusion in my workspace.
This is my system setting: Ubuntu 20.04 + opencv_version 4.2.0 + gcc (Ubuntu 9.3.0-17ubuntu1 ~ 20.04) 9.3.0
I have two questions:
1- which version of Ceres Solver do you suggest 2.0.0 or 1.14.0?
2- at the moment I have 1.14, but yesterday I tried to install ceres solver 2.0.0. Both versions give me some errors during the catkin_make. How can these problems be solved?

 ```peppe@X1:~/catkin_ws$ catkin_make
  Base path: /home/peppe/catkin_ws
  Source space: /home/peppe/catkin_ws/src
  Build space: /home/peppe/catkin_ws/build
  Devel space: /home/peppe/catkin_ws/devel
  Install space: /home/peppe/catkin_ws/install
  ####
  #### Running command: "make cmake_check_build_system" in "/home/peppe/catkin_ws/build"
  ####
  ####
  #### Running command: "make -j12 -l12" in "/home/peppe/catkin_ws/build"
  ####
  [  4%] Built target libGeographiccc
  [  4%] Built target std_msgs_generate_messages_lisp
  [  4%] Built target std_msgs_generate_messages_cpp
  [  4%] Built target std_msgs_generate_messages_py
  [  4%] Built target std_msgs_generate_messages_nodejs
  [  4%] Built target std_msgs_generate_messages_eus
  [  4%] Built target geometry_msgs_generate_messages_nodejs
  [  4%] Built target sensor_msgs_generate_messages_eus
  [  4%] Built target sensor_msgs_generate_messages_cpp
  [  4%] Built target geometry_msgs_generate_messages_eus
  [  6%] Building CXX object VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/chessboard/Chessboard.cc.o
  [  8%] Building CXX object VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/src/intrinsic_calib.cc.o
  [  8%] Building CXX object VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/camera_models/CostFunctionFactory.cc.o
  [  9%] Building CXX object VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/calib/CameraCalibration.cc.o
  [ 10%] Building CXX object VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/src/chessboard/Chessboard.cc.o
  [ 10%] Built target geometry_msgs_generate_messages_lisp
  [ 10%] Built target sensor_msgs_generate_messages_py
  [ 10%] Built target geometry_msgs_generate_messages_py
  [ 10%] Built target sensor_msgs_generate_messages_lisp
  [ 10%] Built target sensor_msgs_generate_messages_nodejs
  [ 10%] Built target geometry_msgs_generate_messages_cpp
  [ 12%] Building CXX object VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/camera_models/PinholeCamera.cc.o
  [ 13%] Building CXX object VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/src/camera_models/EquidistantCamera.cc.o
  [ 15%] Building CXX object VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/src/calib/CameraCalibration.cc.o
  [ 15%] Building CXX object VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/camera_models/CataCamera.cc.o
  [ 19%] Built target global_fusion_node
  [ 20%] Building CXX object VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/camera_models/EquidistantCamera.cc.o
  [ 21%] Building CXX object VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/camera_models/ScaramuzzaCamera.cc.o
  [ 23%] Building CXX object VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/camera_models/PinholeFullCamera.cc.o
  /home/peppe/catkin_ws/src/VINS-Fusion/camera_models/src/calib/CameraCalibration.cc: In member function ‘void camodocal::CameraCalibration::drawResults(std::vector<cv::Mat>&) const’:
  /home/peppe/catkin_ws/src/VINS-Fusion/camera_models/src/calib/CameraCalibration.cc:235:40: error: ‘CV_GRAY2RGB’ was not declared in this scope
    235 |             cv::cvtColor(image, image, CV_GRAY2RGB);
        |                                        ^~~~~~~~~~~
  /home/peppe/catkin_ws/src/VINS-Fusion/camera_models/src/calib/CameraCalibration.cc:253:37: error: ‘CV_AA’ was not declared in this scope; did you mean ‘CV_MSA’?
    253 |                        5, green, 2, CV_AA, drawShiftBits);
        |                                     ^~~~~
        |                                     CV_MSA
  /home/peppe/catkin_ws/src/VINS-Fusion/camera_models/src/calib/CameraCalibration.cc:275:24: error: ‘CV_AA’ was not declared in this scope; did you mean ‘CV_MSA’?
    275 |                     1, CV_AA);
        |                        ^~~~~
        |                        CV_MSA
  /home/peppe/catkin_ws/src/VINS-Fusion/camera_models/src/calib/CameraCalibration.cc: In member function ‘void camodocal::CameraCalibration::drawResults(std::vector<cv::Mat>&) const’:
  /home/peppe/catkin_ws/src/VINS-Fusion/camera_models/src/calib/CameraCalibration.cc:235:40: error: ‘CV_GRAY2RGB’ was not declared in this scope
    235 |             cv::cvtColor(image, image, CV_GRAY2RGB);
        |                                        ^~~~~~~~~~~
  /home/peppe/catkin_ws/src/VINS-Fusion/camera_models/src/calib/CameraCalibration.cc:253:37: error: ‘CV_AA’ was not declared in this scope; did you mean ‘CV_MSA’?
    253 |                        5, green, 2, CV_AA, drawShiftBits);
        |                                     ^~~~~
        |                                     CV_MSA
  /home/peppe/catkin_ws/src/VINS-Fusion/camera_models/src/calib/CameraCalibration.cc:275:24: error: ‘CV_AA’ was not declared in this scope; did you mean ‘CV_MSA’?
    275 |                     1, CV_AA);
[... 151 lines cut ...]
   1592 |         cv::findContours(thresh, contours, hierarchy, CV_RETR_CCOMP, CV_CHAIN_APPROX_SIMPLE);
        |                                                                      ^~~~~~~~~~~~~~~~~~~~~~
  /home/peppe/catkin_ws/src/VINS-Fusion/camera_models/src/chessboard/Chessboard.cc:1597:56: error: ‘CV_THRESH_BINARY_INV’ was not declared in this scope
   1597 |         cv::threshold(black, thresh, threshLevel, 255, CV_THRESH_BINARY_INV);
        |                                                        ^~~~~~~~~~~~~~~~~~~~
  make[2]: *** [VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/build.make:76: VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/calib/CameraCalibration.cc.o] Error 1
  make[2]: *** Waiting for unfinished jobs....
  make[2]: *** [VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/build.make:89: VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/src/calib/CameraCalibration.cc.o] Error 1
  make[2]: *** Waiting for unfinished jobs....
  [ 26%] Built target image_geometry
  [ 30%] Built target cv_bridge
  make[2]: *** [VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/build.make:76: VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/src/chessboard/Chessboard.cc.o] Error 1
  [ 34%] Built target cv_bridge_boost
  make[2]: *** [VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/build.make:63: VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/src/chessboard/Chessboard.cc.o] Error 1
  /home/peppe/catkin_ws/src/VINS-Fusion/camera_models/src/intrinsic_calib.cc: In function ‘int main(int, char**)’:
  /home/peppe/catkin_ws/src/VINS-Fusion/camera_models/src/intrinsic_calib.cc:277:26: error: ‘CV_AA’ was not declared in this scope
    277 |                          CV_AA );
        |                          ^~~~~
  make[2]: *** [VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/build.make:63: VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/src/intrinsic_calib.cc.o] Error 1
  make[1]: *** [CMakeFiles/Makefile2:904: VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/all] Error 2
  make[1]: *** Waiting for unfinished jobs....
  make[1]: *** [CMakeFiles/Makefile2:769: VINS-Fusion/camera_models/CMakeFiles/camera_models.dir/all] Error 2
  make: *** [Makefile:141: all] Error 2
  Invoking "make -j12 -l12" failed
```

MinnDevelopment, 2021-11-24:
Have you tried downgrading to OpenCV version 3?

lnexenl, 2021-11-24:
I am working on Ubuntu 20.04 and met some problems about OpenCV. So I turned to using [Docker support](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion#7-docker-support) provided in the repository, which is easy to deploy and runs well.

WipetheSmlie, 2022-04-01:
One probable reason is your version of OPENCV, vins-fusion is based on opencv3 while your opencv4 may not support some variable's name. Maybe you can try following steps:
1. Locate where is the error from: Based on your terminal feedback, /VINS-Fusion/camera_models/src/chessboard/Chessboard.cc and some other files need to be adapted.
2. Add some head files to these "error" files, according to my own experience: 
    1. In the head file in /camera_model, which is "Chessboard.h", add:
        #include <opencv2/imgproc/types_c.h>
        #include <opencv2/calib3d/balib3d_c.h>

    2. In "CameraCalibration.h", add:
        #include <opencv2/imgproc/types_c.h>
        #include <opencv2/imgproc/imgproc_c.h>

    3. In "loop_fusion/src/VRIEF.h", add:
        #include <opencv2/imgproc/types_c.h>

    4. In "loop_fusion/src/pose_graph.cpp", revise:
        CV_FONT_HERSHEY_SIMPLEX -> cv::FONT_HERSHEY_SIMPLEX

    5. In "vins_estimator/src/feature_tracker.h", add:
        #include <opencv2/highgui.hpp>
        #include <opencv2/cvconfig.h>
        #include <opencv2/imgproc/types_c.h>

    6. In, "vins_estimator/src/KITTIOdomTest.cpp KITTIGPSTest.cpp", revise:
        CV_LOAD_IMAGE_GRAYSCALE -> cv::IMREAD_GRAYSCALE
Above changes are to adapt VINS-FUSION to opencv4
4. Now you would successfully make it, but you need to doublecheck whether there is an opencv2 file in you environment, because in opencv4 it appears be a subfolder of  opencv4

rkuo2000, 2022-04-12:
https://github.com/rkuo2000/VINS-Fusionj
Fixes:
CMakeFiles.txt: set(CMAKE_CXX_FLAGS "-std=c++14")
camera_models/chessboard/Chessboard.cc: #include <opencv2/imgproc/types_c.h>
camera_models/chessboard/Chessboard.cc: cv::CALIB_CB_ADAPTIVE_THRESH, cv::CALIB_CB_NORMALIZE_IMAGE, cv::CALIB_CB_FILTER_QUADS, cv::CALIB_CB_FAST_CHECK
camera_models/src/intrinsic_calib.cc: #include <opencv2/imgproc/imgproc_c.h>
camera_models/src/calib/CameraCalibration.cc: #include <opencv2/imgproc/types_c.h>, #include <opencv2/imgproc/imgproc_c.h>
loop_fusion/src/ThirdParty/DVision/BRIEF.cpp: #include <opencv2/imgproc/types_c.h>
loop_fusion/src/pose_graph.cpp: cv::FONT_HERSHEY_SIMPLEX
loop_fusion/src/keyframe.cpp: cv::FONT_HERSHEY_SIMPLEX

guisoares9, 2022-09-16:
This fork is adapted for OpenCV 4, Ceres 2.1. Tested with ROS Noetic, but it might work on Kinetic. Fork: https://github.com/guisoares9/VINS-Fusion

JaimeParker, 2022-10-26:
opencv version problem, you can revise the `CMakeList.txt` in `camera_models`;

if you installed both opencv 3 and 4, then `find_package(OpenCV REQUIRED)` will probably find opencv 4, so you need to change it to `OpenCV 3 REQUIRED` I guess

rkuo2000, 2022-10-27:
Here's my fork for OpenCV 4.2  (detail code modification are listed in the ReadMe.md)
https://github.com/rkuo2000/VINS-Fusion/

maxpolzin, 2023-01-24:
I added Github Actions to automatically build Docker images of VINS-Fusion running on ROS Noetic, https://github.com/rosblox/ros-vins-fusion. They can be pulled with 
```docker pull ghcr.io/rosblox/ros-vins-fusion:noetic```

Maybe this helps running VINS-Fusion quickly.