HKUST-Aerial-Robotics/VINS-Fusion #169: Units of IMU noise parameters

Kaapeine, 2021-11-29:
In the config files, the IMU noise parameters needed are 'noise standard deviation' and 'bias random work noise standard deviation'. I'm using imu_utils to find the IMU noise parameters and that package outputs 'white noise' and 'bias instability'. 

Is the required 'noise standard deviation' in continuous time or discrete time? Do I need to multiply the value I get from imu_utils with the square root of the sampling rate at which it was calibrated? 

imu_utils outputs bias instability which, as I understand, is quite different from bias random walk. Is there a conversion possible between the two? And which value am I supposed to use in the config file? The continuous value or the discrete one?

Kaapeine, 2022-03-11:
Update: Regardless of which value I tried, I had to tune them manually anyway.

closed 2022-03-11 by Kaapeine

foemre, 2022-08-22:
@Kaapeine what do you mean by manual tuning? Why was kalibr results not good enough? Which imu model are you using? Are the results you obtained satisfactory for you?

Kaapeine, 2022-08-24:
> @Kaapeine what do you mean by manual tuning? Why was kalibr results not good enough? Which imu model are you using? Are the results you obtained satisfactory for you?

Kalibr results were good, I was talking about IMU Calibration using `imu_utils`. At the time of the post, I was using the IMU on the Realsense D435i. The IMU noise values I got from `imu_utils` were too small to be useful practically, with VINS Fusion. By manual tuning, I meant that I had to scale the value 10-100x in order for the system to not fail immediately at start.